iNavFlight/inav #1640: High-speed wireless link - discussion

krzysztofmatula, 2017-05-03:
I'm opening this ticket to discuss possible convenient ways of interfacing iNav UAVs without cables (for flashing, configuring, downloading logs, etc.)

As for today for all the above serial port is used (natively or via on-board USB bridge). 
First thing to determine is what are the actual hard limitations in current hardware? 
921kbps? Is such speed actually achievable and verified? On which hardware?

That's for UART, what then - Bluetooth? The maximum data rate in theory is ~1.3Mbps, but did anyone actually get such speed? (See also: http://wiki.pinguino.cc/index.php/SPP_Bluetooth_Modules)

If not Bluetooth, then maybe WiFi?
We have tiny and nice esp8266. The esp-link firmware (https://github.com/jeelabs/esp-link) seems to support serial port speed up to 460kbps. Still not much and to be confirmed. On the other side we have TCP/IP interface (telnet), convenient for the configurator. 
Also, the UART needn't be the only interface to use. The WiFi part should be quite robust, as it is 802.11 b/g/n and there are reports of >10Mbps speeds achieved.

Any thoughts?

bzdang, 2017-05-04:
A couple of years ago I used XBee data radio modules to link a remote test site to our shop, but have developed a preference for the modules from Synapse Wireless, because each one has a python virtual machine microcontroller which I can reprogram over-the-air. 
There is info here:  [https://solarbotics.com/catalog/wireless/synapse/](url) and at [http://www.synapse-wireless.com/](url)

krzysztofmatula, 2017-05-06:
I've performed some measurements by downloading log file from SPF3 using INAV Configurator:
- on a standard USB connection (115200) - 35kbps actual download rate (bandwidth utilisation: 30%)
- on a Bluetooth link (also at 115200) - 10kbps (bandwidth utilisation: 9% only!!!)

This shows that first thing we'd need is better protocol - kind of "bulk transfer extension", as MWP appears to be far from optimal on high latency links...

Next thing worth considering is data compression. I've measured that typical log file compresses by 50% with zip (44% with zip -1). @martinbudden, what compression rate were you observing with Huffman on https://github.com/iNavFlight/inav/pull/1253 ?

As for increasing UART rates, I've checked the STM32F303 datasheet and it shows that the hard maximum is 9Mbps! Aiming at 921600, we are very far from that limit. And I've also verified that XM-15B Bluetooth module can successfully operate with such rate (however I haven't yet performed real file transfer check with such settings).

Summarizing all the above, aiming at 921600 bps, bulk transfer mode, and some form of compression - transferring 8MB log within 1 minute is achievable!

With that implemented, downloading logs from SD card remotely   (https://github.com/iNavFlight/inav/issues/1336) start to be attractive option.

martinbudden, 2017-05-06:
@krzysztofmatula , I was getting about 30% compression using Huffman.

I've also been doing some work on decreasing the size of the blackbox data, see PR https://github.com/betaflight/betaflight/pull/3000 this gives about a 12% saving in betaflight. I've also been looking at the way data is being stored and I reckon there is scope for better compression in the logs themselves. For betaflight I think I can make the logs maybe 30-40% smaller. I haven't done the calculation for iNav, but it will probably be less, because iNav stores different data to betaflight.

closed 2018-05-27 by stale[bot]