iNavFlight/inav #1641: SP Racing F4 Evo

spursmat, 2017-05-03:
Please could someone add support for this F4 target? It's in latest CF.

Website is a bit ahead of itself...
"Connect a GPS and Compass module for GPS navigation features
Enhanced GPS navigation is available in iNav."
http://seriouslypro.com/spracingf4evo

stronnag, 2017-05-06:
Indeed, vendors might consider providing samples to the development team before insinuating support.

digitalentity, 2017-05-07:
Indeed, having a board will certainly make having support much faster. @hydra?

martinbudden, 2017-05-07:
To be fair, @hydra has sent me a board. Indeed he is one of the few vendors who has sent we a board.

digitalentity, 2017-05-07:
Good to know! Can you add preliminary support for it?

martinbudden, 2017-05-07:
Yes, I plan to. Hopefully sometime during the week.

hydra, 2017-05-08:
I'm happy to donate hardware to iNav developers.  Just pop into the Cleanflight slack channel and drop me a message.

awcherry2, 2017-05-17:
Hello, brand new here and just wanted to chime in that I am looking forward to iNav support for this board as well.  Recently completed a build with it, and wanting to experiment with GPS features.  I am not a programmer, but glad to provide feedback for preliminary support if that would be helpful.

jaahaavi, 2017-06-02:
Any news for adding support for this board?

rokpedler, 2017-06-03:
Hello, I was so happy to hear about INAV. I'm not a racer. I enjoy building quads with as many features as possible then try to get them all working. The idea of getting GPS finally working was really a treat. I even purchased the SP racing F3 NEO and the SP racing F4 EVO to make sure I had hardware to support my INAV projects. I can't understand why these boards, designed by who I think also started Cleanflight,  can't get the support. Yes, I understand that INAV is a fork of Cleanflight. The boards that were shown in the release notes recently supported in CF 2.02 were unheard of to me. Yet nothing for NEO or F4 EVO. I'm sure that I don't understand how all this development stuff works. 
I have 2 SP Racing F3's. I guess I'll have to revert to them.

Sorry for venting. And please forgive any inaccuracies.  I truly appreciate all the work that is done by the developers. This hobby has brought me a wealth of joy because of their strides. Sometimes it confuses me how things happen, or don't happen.
Thank you,
Mike

hydra, 2017-06-03:
@rokpedler sorry, these things sometimes take longer than we'd all like.  We all write this code for fun when we have time​ to do so.  There have been lots of urgent things to do and iNAV support for the new boards has slipped a little.  M having trouble making time for this recently, adding support should be easy enough for anyone who wants to do it, just look at the pin mapping in the CF codebase to get started.

hydra, 2017-06-03:
@digitalentity let me know if you need hardware.  @martinbudden also has hardware but I'm not sure he has time right now.

digitalentity, 2017-06-03:
@hydra having hardware always helps :)
I am a bit distracted recently from INAV development, but now I'm getting
back to adding new features and boards.

or mute
the thread

rokpedler, 2017-06-03:
Thank you gentlemen for your hasty replies. 
You have restored my enthusiasm. I know nothing about coding etc. I'm just a tinkering retiree trying to keep his brain from turning to mush by RC flying and building... and crashing of course. Keep them coming.
Thank you again,
Mike

martinbudden, 2017-06-03:
@rokpedler , the intention is certainly to add support for these boards. It is, as @hydra points out, a question of time. Also to a certain extent a question of knowledge as well - although I know most of the iNav codebase now, my knowledge of the how the timer hardware and pin mappings work is not that good, so for me adding a new target is not entirely straightforward.

@hydra , I've added a PR for the SPRACINGF3NEO board here https://github.com/iNavFlight/inav/pull/1695/files . The `target.c` file, is however just a "best guess" by me - can you review it and correct if necessary?

hydra, 2017-06-03:
@martinbudden yes, thanks for that. I will give it a try on my boards here and check the motor outputs with a scope.

hydra, 2017-06-03:
@digitalentity let me know your address again.

digitalentity, 2017-06-04:
@hydra check you email :smile:

adtinnis, 2017-06-11:
Is there any progress on this.  I believe I have a lot of it done.  I took the work from the Cleanflight/Betaflight target for the SPRACINGF4EVO and moved it over to the iNav tree.  With a few minor changes I believe it is almost there.  The part that I am stuck with is the 'timerHardware' function in 'target.c'.  This seems to be in a significantly different format in iNav to that in Cleanflight.  If somebody can point me at some documentation, I can probably finish this off.

FYI.  I think there needs to be some cleanup on the Cleanflight side as it includes things like OSD in the target, which are not directly supported.

Garogat, 2017-06-11:
Could you push your code please, so we can check it out and help you with it.

adtinnis, 2017-06-11:
I can try.  I am not that familiar with git.  I did a checkout into a feature branch locally.  Committed the changes and then tried to push them into the branch, however I get the error "Permission to iNavFlight/inav.git denied to adtinnis".  Any pointers would help.

Garogat, 2017-06-11:
@adtinnis You can't push directly to INavFlight (most public projects). You have to create your own github fork (use the button on github at inav) and  [push](https://help.github.com/articles/pushing-to-a-remote/) any changes to your personal fork. If you have something finished you can create a pull request at the inav repository.

digitalentity, 2017-06-12:
@adtinnis like @Garogat explained you have to fork the INAV repository and do your changes there. Once you finish you can create a pull request into INAV.

Garogat, 2017-06-12:
I started some timer converting from Cleanflight to iNav.
[Code](https://github.com/Garogat/inav/blob/development/src/main/target/SPRACINGF4EVO/target.c)

nick9one1, 2017-06-14:
I just upgraded to the SP Racing F4 Evo from a Naze 32, assumed iNav would support the board but obviously not there yet. 

I dont have the skills to help with development, but happy to do any testing thats needed.

Garogat, 2017-06-14:
I tired my timer port of cleanflight yesterday, but strangely I wasn't able to detect the usb device correctly. Is that a possible failure by the timers? I can't debug the chip atm as my new st-link will arrive in the next days.

adtinnis, 2017-06-18:
@Garogat @digitalentity I have tested some things and they seem to work.  Have not hooked up motors yet; about to do that.  Receiver works fine.  However there are 2 problems that I can see.  1) Smartport telemetry doesn't work properly (at all in fact).  2) when press 'Save and Reboot' the board doesn't disconnect properly.  Unplugging the board and then plugging in again sorts it out and changes have been saved.  Not a big issue, but not totally right.

I think there may be a wider problem with 'smartport' telemetry.  There are 2 reasons why I believe this:

1) I loaded the BetaFight hex for the SPRACINGF4EVO and smartport works fine.  From this I conclude that my wiring is correct.  However with the iNav target I have built I don't see any telemetry from the flight controller at all.
2) I have a PIXRACER.  I have built a special hex for this where I swapped round the serial ports to enable when is normally UART7.  I swapped this with UART1, which is the WIFI connection on this board.  UART7 has a configurable inverter.  I hard coded this to be on, and used the UART_INVERTER option.  Either way I don't get any telemetry from this target either.  Bear in mind I could have messed this up.  Just another data point.

hydra, 2017-06-18:
on the F4 EVO uart5 uses both pins of the port, so bidirectional/1-wire NOT used.  There is a special telemetry inverter circuit on the F4 EVO and F4 NEO.

check also the serial inversion code/setting in inav

Garogat, 2017-06-20:
@adtinnis I also got mine running now, but how do you connect to the fc? USB-VCP seems buggy to me, so I had to connected via UART1 or ST-Link debugger.

adtinnis, 2017-06-24:
@Garogat I have it working through the serial port.  I don't have a ST-Link debugger and didn't need to connect through a UART.  The problem I have is that the reboot sequence doesn't work.  I have to power the board off and on again (essentially unplug and replug the USB) to get it to reconnect to the configurator.  I am going to open a separate issue on the Smartport telemetry as that doesn't seem to work and I rely on it.  As per my comment above it doesn't work with the PIXRACER either, although that could be down to me.

adtinnis, 2017-06-24:
Scratch the last point about Smartport.  I have it working now.  Had to set the smartport_uart_unidir flag to get it working.  Now it is fine.

nick9one1, 2017-06-25:
It sounds like you're close to releasing this?

nick9one1, 2017-07-04:
how will this be released? Will it be rolled into the next version of iNav, or available as an individual firmware to download?

DzikuVx, 2017-07-04:
@nick9one1 

> how will this be released? Will it be rolled into the next version of iNav, or available as an individual firmware to download?

The first option I'm affraid

adtinnis, 2017-07-04:
I don't think this is ready for merging yet.  I need to do more testing.  There is certainly a problem with the USB connection.  The board does not reboot correctly or at least it does not reconnect to the configuration tool.  The bigger problem seems to be around the motor order.  This may just be my testing, but motor 1 does not work and motors 2-4 are not in the correct order.  When I get some time, I will take a look and try to fix it.

digitalentity, 2017-07-05:
I got a package from @hydra few days ago - will try to find some time to test in the next few days

nick9one1, 2017-07-06:
@adtinnis I've noticed some strange USB behavior in the latest cleanflight with the SPRACINGF4EVO.

- Save and Reboot on the configuration page does not auto reconnect.  The config saves and reboots the board OK, but leaves you on the Cleanflight welcome page. You have to click 'connect'.  

- The CLI also does something similar, if you leave the CLI and navigate to a different area of cleanflight, the board seems to disconnect and take you to the welcome page, meaning you have to click connect again. 

Sounds like it might be related to the USB issues you're experiencing?

spursmat, 2017-07-06:
@nick9one1 I've actually experienced this on older versions of iNav with a Naze32

Merewether, 2017-07-11:
I also jumped early on the proclaimed iNav support for this FC. I have been trying to get Cleanflight to work on my fixed wing aircraft in the meantime and discovered an issue with the servos (Betaflight also has the same issue). I thought that I would make this development team aware of the issue that is detailed in this bug report here:
https://github.com/cleanflight/cleanflight/issues/2792
Although I am new to this framework (and the RC world in general), I am an adequate embedded programmer willing to assist with code development and testing. Let me know if you desire my assistance with this effort.

nick9one1, 2017-07-26:
Has anyone manged to make any progress on this?

awcherry2, 2017-07-26:
I've been wondering about this too.  I hope it is still on people's radar.  Had a few near-misses (our losses) with my build, mostly due to being a new pilot.  I invested in the GPS / MAG almost entirely for the failsafe features; don't want to lose my quad over water or in the woods somewhere.  So it sits on the bench waiting for firmware.

I really don't have a problem waiting on the iNav team, but it is a little frustrating that the SPF4 website still lists the board as having iNav support, even after being out for several months now without it.  Is it time to cut my losses and re-build with a supported board?  Just sucks because I've already bought two SPF4s (bricked the first one due to user error) and I feel vested in this product.  I also want to get back flying though.....

digitalentity, 2017-07-27:
@awcherry2 no worries, we'll get it supported. I have it sitting on my desk - need to find time to hook it up and test at least motors and receiver :smile:

Merewether, 2017-07-27:
FYI, I have also been waiting for this and have been using the Cleanflight in the meantime. I have struggled with one gotcha that you may want to be aware of. The servos do not seem to be correctly mapped to the pins. To get the airplane mode to properly work, the following update to the resource map is required to get the servos on pins 3 through 6 on the connector:
resource SERVO 3 B06
resource SERVO 4 B07
resource SERVO 5 B01
resource SERVO 6 B00

olibye, 2017-08-21:
@adtinnis sorry this is off the EVO topic. You stated: 'Scratch the last point about Smartport. I have it working now. Had to set the smartport_uart_unidir flag to get it working. Now it is fine'

Does that mean you have smart port telemetry working for PIXRACER target. I spent a lot of time trying to get that working with Debug Boards and Digitial Scopes, however I've had no luck.

What did you have to set smartport_uart_unidir to?
Did you ever figure out how to push your changes back to your clone of iNav. I can help you with that if you need. https://help.github.com/articles/fork-a-repo/

matoy, 2017-08-21:
Hi,

+1 for supporting this board :-)
Thank you all.

awcherry2, 2017-08-21:
@digitalentity  Any updates for us?

404Error404, 2017-09-03:
hi, i just bough SPRF4evo on seriously pro,  this board still not working well ? i can't connect anymore to CF conf after trying to upgrade as it noticed when first connect to it. 
im still disapointed. can you help to know what's going on with this board ? 
thanks.

digitalentity, 2017-09-04:
No worries. SPRF4 firmware is in development currently. My board boots and seems to function properly. If you are willing to test it with INAV - I can send you the firmware.

matoy, 2017-09-04:
Hi.
Yes please.
I'm building an Hexa which is still in pieces on my desk but I could start testing some things.

Target0815, 2017-09-04:
Please send me the firmware. I got a sp racing F4 evo today.... Thanks.

digitalentity, 2017-09-05:
Please use this firmware to test: 
[inav_1.7.3_SPRACINGF4EVO.zip](https://github.com/iNavFlight/inav/files/1276163/inav_1.7.3_SPRACINGF4EVO.zip)

matoy, 2017-09-05:
Hi,
it seems that "led" CLI commands are not accepted (i have enabled the feature, saved and rebooted)
it's not working either in the configurator: impossible to "wire" leds.

awcherry2, 2017-09-05:
@matoy Thanks for sending feedback on the LEDs.  I am planning to test INav on my build as well, and will give feedback to this thread as soon as I am able.

In the meantime, would you be willing to share with us more information on your build (configuration, equipment, etc.) and specifically what GPS / MAG are you using?  Are GPS features working for you? Position hold, return to home, etc.?

Thanks so much.  Looking forward to trying this out on mine as well.

digitalentity, 2017-09-06:
Support ready in #2026. Closing.

closed 2017-09-06 by digitalentity