iNavFlight/inav #1643: `OMNIBUSF4` firmware bug.

coloradobum, 2017-05-04:
I've confirmed this on two different flight controller boards and everything works fine in Cleanflight, so I believe it is a bug that was introduced in the INAV fork. I can confirm this behavior in 1.6, 1.6.1 and 1.7.

Using Betaflight and Cleanflight, I can move my sticks on the transmitter and see the channels respond as expected.

When I flash with INAV using the same configs, I don’t see the channels respond in INAV configurator, like I do in Betaflight and Cleanflight.

Here’s what I’m using:

- Flip 32 F4 Omnibus (http://www.readytoflyquads.com/flip-32---f4---omnibus-v3)
- X4R receiver
- Taranis X9D transmitter

I even ordered another flight controller (http://www.2dogrc.com/omnibus-f4-pro-v4-pdb-fc-in-one-smart-audio-support.html) but get the same result.

oleost, 2017-05-04:
Can you share cli dump, and how you wired everything up?

TwoToneEddy, 2017-05-04:
Go to you Rx tab and make sure it's set to default in the mapping tab. I have this board and it seems to default to mapping them all to channel 4

coloradobum, 2017-05-04:
@TwoToneEddy I use the third option, `JR / Spektrum / Graupner` which gives me `TAER1234` a mapping. 

In Cleanflight, if I use `default` (AETR1234) I still see stick movements in the configurator, but they are incorrectly mapped.

In INAV, whether I use `default` or `JR / Spektrum / Graupner` I see no response in the configurator when I move the sticks.

TwoToneEddy, 2017-05-04:
Ok, I assume you're using some kind of serial Rx. If so have you set uart1 to be serial Rx? Do a diff all and I'll try it on my FC when I get a chance.

coloradobum, 2017-05-04:
Hi @oleost

[cli dump](https://www.dropbox.com/s/qv3gqdtxtq9439h/INAV_OMNIBUS_170_DUMP.txt?dl=0)

[receiver wiring](https://www.dropbox.com/s/wsnhdzj0tfhb94h/X4R-SB%20wiring.png?dl=0)

[flight controller wiring](https://www.dropbox.com/s/rp0b3n5tiqia82j/Omnibus%20F4%20wiring.pdf?dl=0)

coloradobum, 2017-05-04:
@TwoToneEddy Yes, correct on the serial Rx. For the Receiver mode I use [this config](https://www.dropbox.com/s/gal7mpisljtrjcu/receiver%20mode.png?dl=0) in INAV.

For this specific flight controller, UART6 is used for SUBS 5V GND (J7 in diagram)  and [this config](https://www.dropbox.com/s/ookitcf2prkfivw/UART6.png?dl=0) is used in INAV.

UART 6 works fine in Cleanflight and Betaflight.

I have tried using UART1, UART3, and UART6 in INAV, but none work successfully. (I believe the UART is tied to the hardware and not firmware, but I tried it anyway. =-) )

TwoToneEddy, 2017-05-04:
Ok, for that connection you need to use uart1 (from my experience, I have the same flight controller with an car mini Rx). Another stupid question, do you have the sbus solder pad soldered?

TwoToneEddy, 2017-05-04:
Oh, another thing, once you've set uart1 to be serial Rx, go back to the configuration tab and make sure it's back on sbus, it seems to flip back to ppm if you change a comm port setting.

coloradobum, 2017-05-04:
@TwoToneEddy I don't have the sbus solder pad soldered per this: [link](https://youtu.be/tKi8Xj9XEKM?t=1m45s)

coloradobum, 2017-05-04:
@TwoToneEddy I claim ignorance here. From my understanding, the UART port number is set on the hardware/chip level and cannot be changed.

For example, If I change Serial Rx to UART1 in Cleanflight and then change the `Receiver Mode` to `serial-based receiver` and then change `Serial Reciever Provide` to `SBUS` it does not work. It only works when I use UART6.

Perhaps my understanding of UART ports is wrong? If so how do "change" UART6 on `J7` on the board to use 'UART1`?

googlebutt, 2017-05-05:
The omnibus f4 v3 (non pro) has a programmable inverter that needs to be enabled in the firmware to work automatically according to the selected serial RX protocol.

They've done it for the pro version, but not for the f4 v3 AIO with SD version from what I can tell.

https://www.rcgroups.com/forums/showthread.php?2792346-Exotic-OMNIBUS

coloradobum, 2017-05-05:
Thank you @googlebutt, I can confirm that the custom firmware `[iNav 1.6.1] OMNIBUS F4 V3 Automatic Inverter on PPM/SBUS (UART6RX)` does indeed work!

I'm not sure of the dev flow on this, but I assume these will be added to the MASTER branch someday.

If I can be of assistance with the process, please let me know how I can help.

In the meantime, I might order and use an F3 version of this board for my GPS build.

Perhaps this issue can be converted to a feature request?

Thanks @oleost, @TwoToneEddy and @googlebutt for your support on this.

digitalentity, 2017-05-05:
@googlebutt are you the author of the required patch? Can you submit a pull request into this repo?

googlebutt, 2017-05-05:
@digitalentity I am not. I would private message Teralift on RCGroups, the author of the thread I linked. He runs support for Omnibus FC's on RCG and either works at myairbot or is in close contact with the engineers.

googlebutt, 2017-05-05:
@digitalentity @coloradobum I forgot to add that the f4 pro doesn't require special inverter care, this is for the f4 AIO v3.

digitalentity, 2017-05-06:
@googlebutt please PM the Teralift and ask if he could provide a patch for INAV. Thanks!

coloradobum, 2017-05-09:
@googlebutt @digitalentity I posted a public reply for a PR: https://www.rcgroups.com/forums/showpost.php?p=37485017&postcount=57

jflyper, 2017-05-10:
Hi. I'm teralift, aka @jflyper ;)

I started to a look at this problem few weeks ago, when I first found out that the OMNIBUS F4 V3 has inverter on a different pin than it's predecessors (actually, the PC10 wasn't used at all). I've created a couple of firmwares for BF and iNav that uses PC8 for inverter on UART6 by merely adding `INVERTER_PIN_UART6 ` def for `OMNIBUSF4SD` (BF, `OMNIBUSF4PRO` for iNav) because serial RX input was switched to UART6; it co-existed with exisiting `INVERTER_PIN_UART1`.

Then, as I started on PR for BF and iNav for this change, I also found out that ASGARD has yet another pin for inverter control on UART6; I had to handle different pin assignments within the same target.

With resource mapping capability of BF3, it came out like this:
https://github.com/betaflight/betaflight/pull/2977

With iNav, something different is required.

moggiex, 2017-05-14:
Its not a bug, it's the board and you need to solder the pads to make the choice between SBUS or PPM, had the same issue myself, see this video [ https://www.youtube.com/watch?v=ZFVIbU4aJvo](https://www.youtube.com/watch?v=ZFVIbU4aJvo) then the board recognises you want SBUS and it then works. Matt

brat002, 2017-05-25:
Anybody has working solution for frsky l9r, which requires inversion?

coloradobum, 2017-06-07:
Thanks for the suggestion @moggiex. Just so others using the OmnibusF4 with a frSKY receiver - know you don't need to do this: https://youtu.be/tKi8Xj9XEKM?t=1m45s

coloradobum, 2017-06-07:
Also, for others, Joshua Bardwell gives the best explanation of F4 boards having only one UART with an inverter that I've seen https://youtu.be/tKi8Xj9XEKM?t=1m45s. FYI: F3 boards have the inverter available on all UART ports.

For others considering a GPS build using Omnibus F4 and FrSKY receivers- save yourself the frustration and use and F3 board for now.

I will add I've been using the custom firmware provided by @jflyper successfully on the F4 board for some time now. Thanks @jflyper!

iconicj, 2017-06-08:
You don't need to modify your receiver. Just use uart1, which has the built in uart inverter.

Also, make sure you're using the right build. Some of them use different pin mapping and take the airbot firmware.

DzikuVx, 2017-08-03:
OK, since we have now OMNIBUSF4V3 target that moves inverter to UART6 we can close this one as it should solve the problem

closed 2017-08-03 by DzikuVx

KnuckleUpFPV, 2018-01-19:
I have the f4 v3 as well. Except my issue is that uart6 doesnt work at all. It wont talk to the gps. It wont talk to my crossfire receiver. It wont respond to my openlrs receiver. Im using uart3 for the scl and sda pads for the mag. Gps on uart1. leaves me with no control link. Ive tried different variations. Even removing the scl and sda wires. 

Used 1.8
and 1.7 variations of firmware. FC has even locked up and once I restarted it. It has completely lost its setup. Goes back to being completely blank. 

When its in my wing, I can get the servos responding to inputs if I dont use the mag and put receiver on uart3. 

However when the wing is in self leveling mode the ailerons respond as if the flight controller is facing the wrong way. If I tip it forward the ailerons move like it was tilted to the left. And opposite if I tip it back. If  i roll the wing they ailerons respond as if its feeling a pitch movement. 

Model tips on the screen correctly with the direction the fc is facing.  If I change the fc orientation to get the servos to respond correctly. The model doesnt match model on screen doesnt match the direction of the FC

This has been driving me crazy. Has anyone experienced anything like this before?

this is the non pdb version.

Listed on amazon as a omnibus f4 v3. Compared it to pictures of the v3 and mine looks identical.