iNavFlight/inav #1644: Heading-hold issues on 1.7.0

krzysztofmatula, 2017-05-04:
I'm having issues with flying in MAG mode (which now should trigger heading-hold mode).

1. The feeling is not good for me. When I try a subtle yaw correction in a stable flight the quad actually resists, keeping the heading. When I move stick further it eventually starts turning, but often more that I wanted. Is it just a feature that can be configured?

2. It happens, that after a turn, when yaw stick is released the quad makes turn back for a significant amount (e.g. 100deg).

See attached log (log 16/16):

[blackbox_log_2017-05-04_190824.TXT](https://github.com/iNavFlight/inav/files/977543/blackbox_log_2017-05-04_190824.TXT)

At 5:40.200 I initiated a turn and the quad changes heading from 139deg to 262deg at 5:43.040. At this point rcCommand[yaw] stays zero, but 0.5s later the heading starts turning back to 153deg for some reason...

3. Why the heading values in the log are in a range 70-430deg instead of 0-360?

My diff:
```
# version
# INAV/SPRACINGF3 1.7.0 May  3 2017 / 17:46:26 (104b66c7)

# resources

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ON_USB

# map

# serial
serial 0 1 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 2 16 57600 57600 57600 115200
serial 31 1 9600 38400 0 115200

# led
led 0 15,15:S:C:2
led 1 14,15:S:CW:2
led 2 9,15:S:CW:2
led 3 8,15:S:CW:2
led 4 7,15:S:CW:2
led 5 6,15:S:CW:2
led 6 1,15:S:CW:2
led 7 0,15:S:C:2
led 8 11,0:N:C:1
led 9 10,0:N:CW:1
led 10 9,0:N:CW:1
led 11 8,0:N:CW:1
led 12 7,0:N:CW:1
led 13 6,0:N:CW:1
led 14 5,0:N:CW:1
led 15 4,0:N:C:1

# color

# mode_color

# aux
aux 0 2 1 900 1200
aux 1 4 1 900 1075
aux 2 4 1 1250 1350
aux 3 3 0 1100 1250
aux 4 3 0 1350 1475
aux 5 3 0 1600 1750
aux 6 3 0 1900 2100
aux 7 9 0 1250 1500
aux 8 8 0 1500 2100
aux 9 11 1 1500 2100
aux 10 12 2 1700 2100
aux 11 18 1 1500 2100

# adjrange

# rxrange

# master
set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 1
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_hardware = MPU6050
set acczero_x = 48
set acczero_y = -31
set acczero_z = 12
set accgain_x = 4082
set accgain_y = 4078
set accgain_z = 4078
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 116
set magzero_y = -54
set magzero_z = -17
set baro_hardware = MS5611
set blackbox_rate_denom = 16
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set nav_extra_arming_safety = OFF
set nav_mc_hover_thr = 1300

# profile
profile 1

set mc_p_pitch = 50
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_i_roll = 40
set mc_d_roll = 20
set dterm_lpf_hz = 30
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45

# 
```

stronnag, 2017-05-05:
3. I only looked at index 16, here the attitude[2] values (decidegrees) range from 0 to 3599.

krzysztofmatula, 2017-05-05:
3.  See heading[yaw], not attitude.

krzysztofmatula, 2017-05-06:
Attaching another log with more yaw turn attempts:
[blackbox_log_2017-05-06_154324.TXT](https://github.com/iNavFlight/inav/files/980996/blackbox_log_2017-05-06_154324.TXT)

Ad.1
At 0:42.070 I start subtle yaw turn - nothing happens until yaw stick reaches "15". Then the rotation starts but the transition from no-turn to turn is not smooth. I believe it starts at turn speed corresponding to yaw=[stick=15], not to yaw=[stick]-[deadzone=15].
Additionally, the dead-zone shall be configurable (5 would be fine for me, 15 is way too much).

Ad.2
At 0:55.609 I release the yaw stick (rcCommand[yaw] goes to zero) and the quad makes a turn back. This happens in both directions.

digitalentity, 2017-05-07:
@krzysztofmatula `heading[yaw]` in blackbox viewer is estimated by the viewer software, it has absolutely nothing to do with what FC acutally sees. Use `attitude[2]`.

Can you take off without using heading hold or position hold modes and do several steady 360-deg yaw turns and provide logs?

krzysztofmatula, 2017-05-07:
@digitalentity, portion of my log attached 2 comments above from 2:49 to 4:17 covers yaw turn tests with no heading-hold enabled. No issues there. There is POSHOLD+ALTHOLD there however. Do you need another log without them?

Xe54, 2017-05-15:
2. Can confirm this behaviour. Quad turns back in both heading-hold and position-hold modes.

krzysztofmatula, 2017-05-16:
Attaching log from different setup (bigger quad, different FC, same software version: 1.7.0):
[LOG00056-tuning3.TXT](https://github.com/iNavFlight/inav/files/1005853/LOG00056-tuning3.TXT)

The same issues can be seen, however they are less prominent.
Best example is at 4:47.800 - slow yaw turn has been finished and then the machine turned back significantly.

Diff:
```
# version
# INAV/SPRACINGF3EVO 1.7.1 May 16 2017 / 18:24:40 (9c1e1618)

# resources

# mixer

# servo

# servo mix

# feature
feature -CURRENT_METER
feature GPS
feature LED_STRIP
feature CHANNEL_FORWARDING
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ON_USB

# map

# serial
serial 1 16 115200 38400 57600 115200
serial 2 2 115200 57600 0 115200

# led
led 0 7,0::CW:1

# color

# mode_color

# aux
aux 0 1 1 900 1200
aux 1 4 1 900 1075
aux 2 4 1 1250 1350
aux 3 3 0 1100 1250
aux 4 3 0 1350 1475
aux 5 3 0 1600 1750
aux 6 3 0 1900 2100
aux 7 9 0 1250 1500
aux 8 8 0 1500 2100
aux 9 11 1 1500 2100
aux 10 12 2 1700 2100
aux 11 18 1 1500 2100

# adjrange

# rxrange

# master
set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 1
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 50
set gyro_notch1_hz = 160
set gyro_notch1_cutoff = 130
set gyro_notch2_hz = 80
set gyro_notch2_cutoff = 50
set acc_hardware = MPU6500
set acczero_x = 7
set acczero_y = 46
set acczero_z = 70
set accgain_y = 4088
set accgain_z = 4072
set align_mag = CW180FLIP
set mag_hardware = HMC5883
set magzero_x = -466
set magzero_y = -164
set magzero_z = -157
set baro_hardware = BMP280
set rssi_scale = 40
set failsafe_procedure = RTH
set nav_extra_arming_safety = OFF
set nav_mc_hover_thr = 1450
set stats_total_time = 407
set stats_total_dist = 53

# profile
profile 1

set mc_p_pitch = 60
set mc_d_pitch = 30
set mc_p_roll = 60
set mc_d_roll = 30
set dterm_lpf_hz = 50
set yaw_lpf_hz = 15
set dterm_notch_hz = 90
set dterm_notch_cutoff = 60
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 30
```

krzysztofmatula, 2017-06-08:
Just a note, that on 1.7.1 the issue is still there.

[blackbox_log_2017-06-08_215428.TXT](https://github.com/iNavFlight/inav/files/1062367/blackbox_log_2017-06-08_215428.TXT)
Most prominent cases (log 5/6): 1:29.600 and 2:08.600.

miradarya, 2017-06-09:
I think the cause for this issue is that getHeadingHoldState returns HEADING_HOLD_DISABLED when tilt angle is greater than the small_angle limit. This then prohibits updating the heading hold target in pidController. According to the blackbox log the SMALL_ANGLE flag is off during the turn, so heading hold target does not get updated, resulting in the quad turning back to the heading hold target that was valid prior to the turn.

I've recently had a different but simular issue with the same underlying cause: When flying real fast in a WP mission (like nav_auto_speed = 1500, which my quad can handle) heading control also drops out intermittently whenever SMALL_ANGLE flag goes off. This is especially noted when turning and accelerating towards a new waypoint. The quad stops turning momentarily and flies sideways until tilt angle gets smaller so that SMALL_ANGLE comes on again. Only then does the quad continue to turn. I can upload a blackbox log in case you're interested.

DzikuVx, 2017-06-09:
@miradarya That is a very interesting observation. Logic to disable HEADING_HOLD when SMALL_ANGLE is crossed is there probably since MultIWii times. I think it was there to "solve" the problem of wrong compass reading when UAV is tilted. 

I think it's time to do something about it finally...

krzysztofmatula, 2017-06-09:
Great, that must be the thing! I haven't spotted the relation with tilt.
This is for the issue 2.

As for case 1, I think the problem is in pid.c and this magic value "15":
```
    if (ABS(rcCommand[YAW]) < 15 && FLIGHT_MODE(HEADING_MODE)) {
        return HEADING_HOLD_ENABLED;
    } else {
        return HEADING_HOLD_UPDATE_HEADING;
    }
```
I think it should be "1" here, not "15", as the YAW input was already processed and yaw_deadband (5 by default) was applied in fc_core.c:
```
rcCommand[YAW] = -getAxisRcCommand(rcData[YAW], currentControlRateProfile->rcYawExpo8, rcControlsConfig()->yaw_deadband);
```

giacomo892, 2017-06-09:
@miradarya wow. Really nice observation. Now I know why my quad during RTH with heading hold activated comes home sideways (which is kind of spectacular lol ) and only after some time (proportional to amount of wind and target speed) it gets the correct heading. 

I thought it was due to fact that I set a low mag hold rate (30) and the quad was reacting slow to automated heading changes.

DzikuVx, 2017-06-09:
@krzysztofmatula once again, reasoning for this must be looked for somewhere in MultiWii...

Either way, let's check it one step at a time, #1780 keeps maghold event during high tilt angle.
I will try to test it on my 6" quad this weekend, but if anyone else want's to give it a try, be my guest!

digitalentity, 2017-07-16:
I believe this issue is fixed. Heading hold is now working at angles up to maximum allowed inclination and don't have any additional deadband on the stick. Feel free to re-open if issue presists.

closed 2017-07-16 by digitalentity