iNavFlight/inav #1645: Change request: Use mode box names instead of numerical identifiers in CLI dumps

krzysztofmatula, 2017-05-05:
The set of supported flight modes tend to change from time to time in inav... This sometimes results in different modes being actually configured by 'aux' settings when old diff is applied to new release.

I propose to use textual names for flight modes instead of mode identifiers.

Current dump:
```
# aux
aux 0 2 1 900 1200
aux 1 4 1 900 1075
aux 2 4 1 1250 1350
aux 3 3 0 1100 1250
aux 4 3 0 1350 1475
aux 5 3 0 1600 1750
aux 6 3 0 1900 2100
aux 7 9 0 1250 1500
aux 8 8 0 1500 2100
aux 9 11 1 1500 2100
aux 10 12 2 1700 2100
aux 11 18 1 1500 2100
```

Proposed:
```
# aux
aux 0 HORIZON 1 900 1200
aux 1 HEADINGHOLD 1 900 1075
aux 2 HEADINGHOLD 1 1250 1350
aux 3 NAVALTHOLD 0 1100 1250
aux 4 NAVALTHOLD 0 1350 1475
aux 5 NAVALTHOLD 0 1600 1750
aux 6 NAVALTHOLD 0 1900 2100
aux 7 NAVPOSHOLD 0 1250 1500
aux 8 NAVRTH 0 1500 2100
aux 9 BEEPERON 1 1500 2100
aux 10 LEDLOW 2 1700 2100
aux 11 FAILSAFE 1 1500 2100
```

martinbudden, 2017-05-05:
I like this idea.

digitalentity, 2017-05-05:
I like the idea!

closed 2018-05-27 by stale[bot]