iNavFlight/inav #1646: Omnibus F4 v3 baro is red

brat002, 2017-05-05:
Fresh Inav 1.7.0 on genuine Omnibus F4 v3.

`# dump

# version
# INAV/OMNIBUSF4 1.7.0 May  3 2017 / 11:22:34 (104b66c)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature VBAT
feature CURRENT_METER
feature BLACKBOX

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
[... 511 lines cut ...]
C09: FREE 
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: FREE 
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE`

Is my Omnibus F4 faulty?

brat002, 2017-05-06:
I had flashed OMNIBUSF4 target. OMNIBUS F4PRO works well.

oleost, 2017-05-06:
Glad you sorted it out.

closed 2017-05-06 by oleost