iNavFlight/inav #1650: LAUNCH_MODE improvements

brat002, 2017-05-06:
Current implementation of LAUNCH_MODE is not safe. Instead of activate climbing after delay LAUNCH_MODE should set elevator/elevons up as only LAUNCH_MODE was activated.
Typical safe procedure of launch:
* activate launch mode (may be good option to allow activate launch mode after power on)
* activate arm
* set desired throttle level for safe take off (motor should spin at this level to be ready to make required thrust)
* throw plane by wing tip or fuselage

Main goal of these changes - flight controller have to be ready to stabilize plane in process of throw and after it. Activate climbing after throw sounds good, but mostly is not usable by beginners.

Please, improve it.

digitalentity, 2017-05-07:
@brat002 climb angle is set from the moment LAUNCH_MODE is activated. FC is stabilizing the plaine during the throw and after it. After motor starts the integrator is activated to compensate for imperfect trim.

Launch mode starts motors with a delay for a safer throw or a catapult launch where propeller can't spin until the aircraft is clear from the launch site. 

If you want motor to spin right away - don't use launch mode.

brat002, 2017-05-07:
Yesterday I had bad experience with launch mode. The main problem - plane did not climb first several seconds, but after 15-20 meters started climbing. Plane was throw by wing tip.
Throttle idle speed set to 2000.

Soph, 2017-05-07:
@brat002 what works for me when using it with a wing: Hold it in the same way as I would normally throw, do a shake until motor starts and then throw with running motor. Works perfect.

brat002, 2017-05-08:
But imagine if shake was accidentally and your fingers were near with prop.
Much safer to directly control motor with throttle stick, but for catapult
launch use special sub-mode like "delayed motor start".

digitalentity, 2017-05-08:
What you request is unrelated to `NAV LAUNCH`. The point of launch assistance is making FC control throttle and pitch to allow the pilot concentrate on throwing the airplane correctly without holding the TX and bothering about the sticks.

Btw, nobody is forcing you to use launch assistance. You can always use ANGLE mode + custom mix on your TX to request a certain climb angle and control throttle manually with your TX.

digitalentity, 2017-05-08:
I think this issue is irrelevant and can be close.

brat002, 2017-05-08:
My point is to control motor directly with throttle stick and activate climbing as only launch_mode was activated, but not after shake plane. Reason to do that:
* it is easy to stop motor in case of interrupt launch - just pull down the throttle stick. When you hold plane with right hand - second hand is always on throttle stick.
* human will control better necessary speed in every condition. In case of strong wind - launch speed may differ.

digitalentity, 2017-05-08:
Let me clarify LAUNCH mode logic:

1. It keeps nose up **always**, right from the moment of mode activation and until sequence finished. Only starting motors at predefined power is linked to detecting actual motion of the plane
2. Pilot is free to kill motors with disarm switch at any time
3. Pilot is free to take over full control at any time by moving the right stick

Once more - what you so desire can be achieved without launch mode: use ANGLE mode + custom mix on your TX to pitch up. You'll get manual control over throttle and speed. You'll be free to kill motors by pulling throttle stick down. You'll be able to set throttle to any level you want and motor will be working before the throw.

closed 2017-05-08 by digitalentity