iNavFlight/inav #1655: Feature Request: Servo output logging in blackbox

miberlin, 2017-05-07:
Hey Guys,

coming from quads I am used to checking the motor outputs in my blackbox logs when I feel like one of the motors is broken. If I see that the quad tilts to one side even though the motor on that side is ramping up rapidly, I know that this motor is broken.

The same would apply to a plane - If the plane spins to one side and iNav tries to stabilize using the servos I could see in the blackbox file, which servo might be broken since this servo is going to get a higher signal than the others.

Is there any way we could get the servo output into blackbox?

Thanks a lot!
mi

DzikuVx, 2017-05-17:
@miberlin PID_sum is more less equivalent to servo output

fahrvergnuugen, 2019-03-09:
I would like this feature too. It would be nice to know what PWM value was sent to each output. Getting PID SUM isn't as transparent. I'm trying to see how much the FC is fighting to keep the plane flying flat and level on a calm day to work out elevon warping issues.

hali9, 2019-03-11:
https://github.com/iNavFlight/inav/issues/1869

shellixyz, 2019-03-11:
@hali9 Your feature request have been closed by accident I will leave this one open instead if you don't mind.

hali9, 2019-03-28:
This can be closed

closed 2019-03-28 by giacomo892