iNavFlight/inav #1656: 5" GPS preset mode selected gyros and sticks do not respond

autobotix, 2017-05-07:
Hi
My first time using Inav, I tried selecting 5" GPS preset mode and find that the gyro's do not respond to and movement. The interface shows everything operating as expected, but a hand test shows no response from the gyro's and the stick input. Switching back to Racer preset resolves the issue.

oleost, 2017-05-07:
Are you using latest version of INAV configurator?

8. mai 2017 12.16 a.m.

autobotix, 2017-05-08:
yes I believe so
2017-05-08 @ 11:21:32 -- Running - OS: MacOS, Chrome: 57.0.2987.133, Configurator: 1.7.0

oleost, 2017-05-08:
I tried both INAV 1.6.1 and INAV 1.7 on SPRACINGF3 board on my desk.

I dont have any receiver hooked up, but on both firmware GYRO works as expected.

autobotix, 2017-05-08:
interesting
I can reaily switch between modes and in 5" gps there is no gyor or receiver control whatsoever, like all pids were at 0
everything shows and reacts as expected in the gui.
Any idea what to check?

oleost, 2017-05-08:
Have you tried reflash with clean erase?

Are you using the spracingf3 target?

Provide bootlog and tasks and status dump

autobotix, 2017-05-08:
can you please supply cli commands
yes using spracingf3 target.
Let me reiterate, 
it works correctly if i use the "racer" preset this only seems to be an issue in 5" GPS
I have reflashed numerous times, but I will again to verify 100%

oleost, 2017-05-08:
'bootlog' 'tasks' and 'status' ...

DzikuVx, 2017-05-17:
@autobotix  please provide 'bootlog' 'tasks' and 'status' as provide more information on hardware

DzikuVx, 2017-06-09:
Since there is no activity in this support ticket, I'm closing it

closed 2017-06-09 by DzikuVx