iNavFlight/inav #1662: INAV 1.6 not using motor for RTH

Paddy79, 2017-05-11:
I have iNAV 1.6 in my glider on a SPRACING F3 Acro. ESC is calibrated but it is not using the motor for RTH and so against wind it can´t make it home. I am new to iNAV so maybe I have just made a mistake somewhere. I can post my settings if that helps to find the problem.

oleost, 2017-05-11:
It should use motor.

Provide cli dump and an blackbox recording of it.

oleost, 2017-05-11:
Also official 1.7 is out. Please update to latest version

Paddy79, 2017-05-11:
# version
# INAV/SPRACINGF3 1.6.1 Mar  9 2017 / 00:47:57 (6c878ae)

# resources

# mixer
mixer AIRPLANE

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 -100 -1
servo 4 1000 2000 1500 90 90 -100 -1
servo 5 1000 2000 1500 90 90 -100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
beeper ACTION_FAIL
beeper READY_BEEP
[... 391 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 15
set yaw_rate = 9
set tpa_rate = 0
set tpa_breakpoint = 1500

#

oleost, 2017-05-11:
You don't have RTH failsafe set failsafe_procedure = SET-THR.

Please read the Wiki. And use forum for further support.

closed 2017-05-11 by oleost