iNavFlight/inav #1663: Position Hold problem

adtocco, 2017-05-11:
Hello, it's first time i use inav, i have spracing f3 FC and inav 1.7.

I try to use position Hold ( Gps fix is good, magnetometer too in GUI)

When i activate position hold in flight, the quad keep his position only if he is oriented to the North ( heading = 0° ). When the quad is oriented to West East or South, he fly away.

anyone have already solved this problem ?

stronnag, 2017-05-11:
If you are using the onboard mag, it is probably interference. Use an external mag. The static UI is useless to debug these issues. Please post a CLI `diff` and a blackbox recording.

DzikuVx, 2017-05-23:
This issue is  hardware related just like @stronnag  suggested. MAG not working like expected or too close too power lines. Since not additional information was provieded, I'm closing this one.
@adtocco if you have any hint that this is software related, plaese inform

closed 2017-05-23 by DzikuVx

tomla1963, 2017-07-11:
Hi
Sorry hijack this issue.
I have also spracing f3 fc with inav version1.7.1 with same issue.
From banggood "F3 Flight Controller Deluxe 10 DOF for Multirotor Racing"
Althold working fine.
PosHold working fine when heading north but not heading south.
It could still be onboard mag that are causing the problem.
I have tried to tried several position and shielding with no luck.
Here is my cli diff.

# version
# INAV/SPRACINGF3 1.7.1 Jun  3 2017 / 08:13:23 (48e1feff)

# diff

# version
# INAV/SPRACINGF3 1.7.1 Jun  3 2017 / 08:13:23 (48e1feff)

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature GPS
feature RX_PARALLEL_PWM
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 1 2 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100
aux 1 1 0 1300 2100
aux 2 3 0 1600 2100
aux 3 9 0 1575 2100

# adjrange

# rxrange

# master
set acc_hardware = MPU6050
set acczero_x = 37
set acczero_y = 22
set acczero_z = -93
set accgain_x = 4082
set accgain_y = 4075
set accgain_z = 4055
set mag_hardware = HMC5883
set mag_declination = 580
set magzero_x = -69
set magzero_y = 108
set magzero_z = 58
set baro_hardware = MS5611
set min_throttle = 1161

# profile
profile 1

set mc_i_pitch = 20
set mc_i_roll = 20
set roll_rate = 30
set pitch_rate = 30

Regards
tom

giacomo892, 2017-07-11:
Onboard mag is going to fail. Remove it and use an external one.