iNavFlight/inav #1664: OSD: GPS speed display corrupted. Omnibus F4 Pro V2 Fixed Wing

Martijn666, 2017-05-11:
Dear Devs,

I noticed on the latest 1.7 that the GPS speed icon for km/h is wrong. Also in the summary screen, the number doesn't correspond with the max velocity. See https://youtu.be/etGFGMyuhWY
Setup: S800 Sky Shadow, INAV/OMNIBUSF4PRO 1.7.0 May  3 2017 / 11:23:36 (104b66c)

flounderscore, 2017-05-11:
That must have been my mistake. I did some chances while implementing
imperial units, but never had time to test them. There might also be a
mismatch between the character constants and the font.

Thanks for reporting it.

digitalentity, 2017-05-15:
@flounderscore I believe fix should be fairly simple, no?

flounderscore, 2017-05-15:
I'm getting the hardware for a new test setup (hopefully) today. I will
have a look.

flounderscore, 2017-05-16:
@Martijn666 If you have a chance, can you (or someone) else test #1688? Should be working now, but it's going to be a while before I can test and fly it.

Martijn666, 2017-05-16:
It will take me some time, will try to test tomorrow. Thanks.

flounderscore, 2017-05-16:
Could you please also verify that distance and altitude alarms are working as expected?

Martijn666, 2017-05-16:
I'll compile and flash now, testing will take place tomorrow afternoon. I'll post a dvr recording.

Martijn666, 2017-05-16:
I have difficulties compiling. Can someone please compile TARGET=OMNIBUSF4PRO including #1688 ?

flounderscore, 2017-05-16:
I'll upload a hex for you tonight (1h or two).

Martijn666, 2017-05-16:
Thanks @flounderscore 

[image]
That's all I get.

flounderscore, 2017-05-16:
@Martijn666 You can get the hex from: https://ufile.io/4r3ep It's built form yesterday's development/HEAD

The checksum is (in case you're paranoid)
MD5 (./obj/main/inav_OMNIBUSF4PRO.elf) = 4b4b1edfd4d5bd61e4052b1c7bd978e2

Thanks for testing.

Martijn666, 2017-05-16:
Just my luck. Destroyed the bootloader button. I have another f4pro in the mail, expecting tomorrow. Stay tuned.

flounderscore, 2017-05-17:
Flight tested today (in imperial mode) and max speed and distance from home was correct. Not sure about distance travelled (quite a large number in "ft", but probably correct).

digitalentity, 2017-05-18:
Fix merged. Closing

digitalentity, 2017-05-18:
@flounderscore thanks!

PR #1688 (2017-05-18): Fixed OSD symbols for velocities and stats calculation
Attempt to fix #1664. Bench tested for the correct symbols, but couldn't test velocity/stats.
closed 2017-05-18 by digitalentity