iNavFlight/inav #1666: Board freeze in combination with Hott telemetry

heliflieger, 2017-05-12:
Hi,

I have a freeze problem
Hardware: SP F3 EVO, Graupner Hott GR-16 receiver
Software INAV 1.7.0

If I connect the GR-16 Hott telemetry with the diode to the FC (Flight Controller), the controller freeze.
All LEDs on the FC freeze, the display on the I2C port are also freeze. The freeze does not came immediately. It came on various times. For e.g. if the GPS get a fix.

If I disconnect the telemetry the board doesn't freeze.

Any ideas what I can do to solve this problem.

digitalentity, 2017-05-13:
Please share your configuration dump.

heliflieger, 2017-05-13:
here are the dump:

# version
# INAV/SPRACINGF3EVO 1.7.0 May  3 2017 / 11:13:56 (104b66c)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature LED_STRIP
feature DASHBOARD
feature BLACKBOX
feature TRANSPONDER
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
[... 404 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

digitalentity, 2017-05-13:
Thanks. Will investigate

digitalentity, 2017-05-13:
Unable to reproduce on my boards, but found a possible bug in the code. Can you please test the attached firmware?

[inav_1.7.1_SPRACINGF3EVO.hex.gz](https://github.com/iNavFlight/inav/files/998379/inav_1.7.1_SPRACINGF3EVO.hex.gz)

heliflieger, 2017-05-13:
Better but not all ok.
The board doesn't freeze anymore, but the telemetry data will toggle in unarmed state.
So for the most time there will no telemetry data send to the transmitter.
During armed state there are no toggle. The telemetry data will be send during the full time.

digitalentity, 2017-05-13:
Ok, I need to set up my test rig again and see what's going on on the wire. It's probably GPS message that's causing issues...

digitalentity, 2017-05-16:
Should be fixed by https://github.com/iNavFlight/inav/pull/1691

digitalentity, 2017-05-16:
Closing. Feel free to re-open if bug persists.

closed 2017-05-16 by digitalentity

heliflieger, 2017-05-16:
Can you please build me a test version?

digitalentity, 2017-05-19:
@heliflieger sure:

[inav_1.7.1_SPRACINGF3EVO.hex.gz](https://github.com/iNavFlight/inav/files/1014042/inav_1.7.1_SPRACINGF3EVO.hex.gz)