iNavFlight/inav #1668: CLI current scale max should be much larger than 1000

skaman82, 2017-05-12:
With my current setup I need to input a current scale of 2674. Its not possible because the max value is 1000.

DzikuVx, 2017-05-12:
I see that code say -10000 to 10000

```{ "current_meter_scale",        VAR_INT16  | MASTER_VALUE, .config.minmax = { -10000,  10000 }, PG_BATTERY_CONFIG, offsetof(batteryConfig_t, currentMeterScale) },
   ```

Anybody knows why scale can be lower than 0?

skaman82, 2017-05-12:
My bad, overseen a zero :D But it is still 1000 in configurator

DzikuVx, 2017-05-12:
@skaman82 not anymore

krzysztofmatula, 2017-05-15:
> Anybody knows why scale can be lower than 0?

Didn't try this yet, but it is for sensors that report lower voltage as the current increases. Like for example ACS756KCA (which may be used both ways, with actually the reverse way being more attractive as it would guarantee the voltage will not exceed 3.3V, which is a limit for most FCs).

DzikuVx, 2017-05-15:
OK, closing this one

closed 2017-05-15 by DzikuVx