iNavFlight/inav #1669: Omnibus F4 V3 - RGB Led doesn't work

DangerD256, 2017-05-13:
I have this one:
http://shop.myairbot.com/index.php/flight-control/cleanflight-baseflight/omnibusf4v3.html

RGB Led works on Betaflight fine.

flounderscore, 2017-05-13:
Did you use the led connector or PWM5/6?

DangerD256, 2017-05-13:
Led connector, works well on betaflight

flounderscore, 2017-05-13:
There were some timer conflicts on earlier revisions of the board that where fixed by moving the led to a different port, if I remember correctly. Some of the devs might chime in.

DangerD256, 2017-05-13:
Need to move it back =)

jstremmler, 2017-05-19:
In betaflight LED_strip is on B6 for OmnibusF4 Pro V3 and (mandatory) on A8 for OmnibusF4 Pro V2.
In iNav it is by default on A1 if you switch feature LED on. So I would try A1 (=  PWM Motor5) for the OmnibusF4 Pro V3

closed 2018-05-27 by stale[bot]

modellbobby, 2018-08-10:
Anything here? I cannot make it work on PWM5