iNavFlight/inav #1670: Transfer BB logs extremly slow

DerElch87, 2017-05-13:
Hi there,
I noticed that the BB log transfer is extremly slow on INav compared to Betaflight.
5900 KB takes to transfer with:
INav 1,7 ~ 29min
BF 3.1.7 ~ 10min

I tested this on two different quads.
Both SP Racing F3 with 8MB onboard flash
Same USB cable used for both.

What I see at the INav BB transfer:
 - Sometimes the transfer seems to hang for more than 30s (the size of the log doesn't increase for a certain period)
 - Sometimes the size only increases by 1KB per 30s
 - When it is transfering "normally", so not in one of these hang times, it transfers about 4,3KB/s

Both FC are mounted so I can't look at the onboard flash chips if there is a difference.
Somebody else experienced the same?

stronnag, 2017-05-13:
If you compile with `USE_FLASH_TOOLS` and then use something like https://github.com/stronnag/mwptools/blob/master/samples/flash_dump.rb to down the data, you get (iirc, I'll record the time later) betaflight like timings. At least this is my mitigation for slow bb downloads.

oleost, 2017-05-13:
If INAV configurator is not the active window Chrome will automaticly
throttle it down to 1% of cpu power.

http://www.askvg.com/how-to-enable-disable-background-tabs-throttling-feature-in-google-chrome-and-opera-web-browsers/

DerElch87, 2017-05-13:
@stronnag 
Thanks, maybe I'll try that one time
@oleost 
This can't be the reason, because then it would be the same for both, INav and BF.

Are there any huge differences between both and why there is such a difference?

oleost, 2017-05-13:
Unsure how you tested this and if your was not aware of this "issue" you
could accendtiatly have opened another window.

DerElch87, 2017-05-13:
To be 100% sure I'll test it the next days again with the Configurater in foreground.
I'll give feedback soon

stronnag, 2017-05-13:
So, let's hack cli.c to allow baud rates > 115200 for MSP #1673 and apropos #1464 
````
$ flash_dump.rb -b 921600
Size = 1857536
read 1857536 / 1857536 100%    0s
Got 1857536 bytes in 21.1s 87853.2 b/s
Exiting
````
Nice, very nice compared to 160s at 115200 baud.

DerElch87, 2017-05-13:
@oleost 
Just to give feedback again.
Inav Configurator always in foreground.
8192KB took me 33min.

@stronnag 
Is there anywhere a "how to" or tutorial how to compile something like INav?
To be honest I have no idea how I can do something like USE_FLASH_TOOLS...

The only programming experience I have is Arduino, and this is simple...

stronnag, 2017-05-13:
Compared to an extrapolated 95s ( USE_FLASH_TOOLS, 921600 baud :), tedious.

Compiling is easy (easier than arduino in my experience). There are developer guides for major platforms in the iNav developer documentation. These tend to be out-dated in terms of latest compilers. 

At its simplest it involves installing the cross-compiler and typing`make`. For Linux, for a modern, up-to date distro, the distro's cross compiler will usually be adequate (Arch, Ubuntu 17.04, Fedora 25); for Windows or MacOS it may be a bit harder.

Assuming you can get the cross compiler installed, then to build with USE_FLASH_TOOLS for a given target (let's assume SPRACINGF3):
````
make TARGET=SPRACINGF3 OPTIONS=USE_FLASH_TOOLS=1
````
The resultant hex file will `obj/inav_1.7.1_SPRACINGF3.hex` (dev) or `obj/inav_1.7.0_SPRACINGF3.hex` (master)

DerElch87, 2017-05-14:
@stronnag 
Thank you very much, I'll really appreciate it.

But then there is still the open point:
Why is blackbox download with INav so much slower compared to Betaflight?
Bug?
Feature missing?

digitalentity, 2017-05-14:
We don't have full MSP jumbo frames support. Betaflight uses MSP jumbo frames to transfer logs and thus do it much faster (almost as fast as `USE_FLASH_TOOLS`).

DerElch87, 2017-05-14:
Thank you @digitalentity 
At least I see now that there is a reason for that :)

Is it possible that INav also supports the MSP jumbo frames in future?

By the way, great work here with INav.
I my good old F450 is now flying better than ever.
From MultiWii --> OpenPilot --> LibrePilot --> Cleanflight --> INav
Keeping up the good work!

digitalentity, 2017-05-14:
I think we have many things already in place, it's just a matter of putting them together. I'm not sure about support on Configurator end. @DzikuVx ?

DzikuVx, 2017-05-14:
Should be fairly simple I think

fiam, 2017-09-08:
Should we consider this fixed on 1.7.3? MSPv2 should make flash downloads way faster.

closed 2017-09-09 by digitalentity