iNavFlight/inav #1674: X220 not working

ianaz, 2017-05-13:
I got an Eachine x220 wizard which I soldered a GPS module, everything works in terms of communication and the gps is receiving all the data.
When I try to fly the drone on ACRO mode, all motors are spinning correctly, which is good.
When I try to fly the drone on ANGLE or HORIZONTAL, all motors are spinning but keep increasing the velocity even if I have throttle at 1200.  I tried to re-flash the drone with Inav and do only the basic setup without GPS configuration, same issue.

Also, the voltage sensor is not working at all.

I was flying well with CleanFlight and BetaFlight before I moved to Inav, which never worked on angle mode.

Flight controller is a F3 and gps is a UBLOX M8N

digitalentity, 2017-05-14:
Regarding motors - did you try actually *flying* the drone? Does voltage sensor work on Betaflight? Which version of Betaflight? Also please consider providing config dump of your setup.

DzikuVx, 2017-06-09:
Since there was no activity is a problem that looks like a support ticket, I'm closing it. @ianaz if anything new appears feel free to reopen or create a new ticket. Thank you

closed 2017-06-09 by DzikuVx

rinyheuvelman, 2018-05-02:
First update your f3  racing fc board to latest betaflight, see https://www.youtube.com/watch?v=e7dGCZMJ6SA
that fixed it for me. 
Make sure your rx is plugged the right way
for ibus:horizontal top three pins on the right, to boardconnection IO1
for ppm: three vertical pins on left bottomside to boardconnection IO2
connection see http://nathan.vertile.com/assets/images/blog/fs_i6/500px-FlySky_IA6B_pinout-f89eed4f.jpg
FC: 
[SPRacingF3-Manual-latest.pdf](https://github.com/betaflight/betaflight/files/1966781/SPRacingF3-Manual-latest.pdf)
[FS-i6-MANUAL-CN-EN-20150601.pdf](https://github.com/betaflight/betaflight/files/1966783/FS-i6-MANUAL-CN-EN-20150601.pdf)