iNavFlight/inav #1681: Feature Request: Add toggle for nav_rth_allow_landing in GUI (and default OFF for Fixed Wing)

moggiex, 2017-05-14:
Howdy,

Just been out and flown V1.7.0 in a flying wing. Worked fantastic this time around, YAY! 

However the model when returning to home wanted to ditch itself. I was expecting the model to loiter rather than attempting to land. 

As such, can the option "nav_rth_allow_landing" which is set to ON by default for fixed wing models be added to the GUI so that the user has the option to change this value more easily (ie not in the GUI after finding out it even exists).

Also as a secondary suggestion, being used to other FC setups, for fixed wing, the typical action is to RTL and then loiter. Would be it an idea to have this set to OFF for fixed wing models?

Matt 

PS. V1.7.0 did try aggressively try and plant the model in the ground when it returned, but I was aware that may happen and is covered in another thread.

Redshifft, 2017-05-15:
It wold be interesting to know how many use this for fixed wing first ;)

IMHO it has limited use as few of us have enough open landing area for the aeroplane to have a safe arrival every time by itself.

Default auto-land should certainly be off for sure.

oleost, 2017-05-15:
Today there is common auto land setting for both fw and mr. So as it is now it must be default turned on, without a doubt.

Well, it all depends what your are using RTH for.

My use case is for Failsafe when flying. 
First scenario is I fly out of rang,e it would turn home and as soon I have proper control I would take over control and fly it myself.

Second scenario is that something goes wrong with my ecuipment, example loose servo cable between RX and FC. In this case I would have my plane attempt to bring the aircraft down, instead of loitering 50-100meters up in the air until lipo is discharged.

Redshifft, 2017-05-15:
Original poster was talking RTH not Failsafe ??

MR and FW are different animals when considering landing precision and approach trajectories etc, so a common auto landing "across" iNAV is just not the best idea.

moggiex, 2017-05-15:
Howdy gents,

For clarity, RTH is what I was referring to. For a "fail safe" I would also assume the same. 

When a Fixed Wing model returns, the model **should loiter**, not land.

To make my point, as I've used all of these and this is their default condition under failsafe or RTH, the model loiters when it returns. It's "an industry norm".

Note: Obviously these can be altered using settings, but the default for ALL of these devices is the same out of the box.

EagleTree Vector
----
Loiters on it's return, http://eagletreesystems.com/Manuals/vector.pdf (section 5.1.12.2 on page 57)

Arduplane
----
Loiters on it's return, http://ardupilot.org/plane/docs/rtl-mode.html

QGroundControl with PX4 firmware
----
Loiters on it's return, https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/en/SetupView/FlightModes.html

FY-41AP Lite & Pro
----
Loiters on it's return, http://www.feiyu-tech.com/index.php/Product/detail/pr_id/58.html

There are other devices as well, basically none of them for Fixed Wing models land on their return by default. Hence the suggestion not to land, instead loiter by default for Fixed Wing.

And just to add another scenario, many fixed wing FPV pilots will launch their model line of sight, hit RTH because they're paranoid about it working, let it circle above their heads, get seated and then go on a jolly with the model.

Hope the extra background on this helps,

Matt

digitalentity, 2017-05-15:
Sounds reasonable. We probably need this in presets (will require corresponding MSP API for this)

moggiex, 2017-05-15:
Ah ha, adding this as a preset for the fixed wing models would be a great idea. Good thinking @digitalentity!

Matt

PS. Having it as a sub, toggle option in the GUI would also be nice :)

digitalentity, 2017-05-15:
At the moment we don't have MSP API for fixed-wing navigation parameters - we need to implement one first :dagger:

moggiex, 2017-05-15:
I have no idea what the MSP API is, so I'll let you deal with that one :)

moggiex, 2017-05-16:
Howdy, can I also add "**nav_rth_climb_first**" to **OFF** as part of this request please @digitalentity.

That var which defaults to ON explains why the model would carry on in whatever direction it was flying in to get to the defined height and then turn as seen in this video [https://www.youtube.com/watch?v=Mc3-FgqE8dg](https://www.youtube.com/watch?v=Mc3-FgqE8dg)

Again, using the other flight controllers above as a reference, they all turn first and then climb when RTH is activated. 

On a windy day we may go past the value of **nav_rth_abort_threshold** before we hit the return to home height, especially if that height is say 100M or even more. Also the act of facing the model back towards the home location is likely to improve a FPV video link, either from the orientation of the model and/or the model getting closer to the receiver.

I hope that makes sense. 

Matt 

PS. Changing **nav_rth_altitude** to a more apt value of say 3000 for 30M would also be suggested. 10M is not high enough to clear most trees. But I feel I'm pushing my luck with suggestions ;)

flounderscore, 2017-05-17:
Is there a way to distinguish RTH flight mode from RTH failsafe? The latter should have auto-land for fixed wing, the first shouldn't IMHO.

oleost, 2017-05-17:
Mhm, that was my initial proposal. Have land on failsafe initiated RTH.

DzikuVx, 2017-05-17:
Part of work done in #1698 so I suppose we can close this one.

@moggiex all Presets changes please put in Configurator project ( https://github.com/iNavFlight/inav-configurator )

Closing this one

closed 2017-05-17 by DzikuVx