iNavFlight/inav #1682: Is mag_calibration_time working ?

grmis-fr, 2017-05-14:
Hi. I am using inav 1.6.1 on a quad, and I would like to increase the magnetometer calibration time. However, whatever the value I set to mag_calibration_time (I tried 60 and 120), the still calibration stops after 30 seconds. Did I miss something ?

krzysztofmatula, 2017-05-15:
Looking at the code it seems it should be respected. Supported range is 30..120s.
How do you know it stops calibrating? Does it stop blinking and beeps?

grmis-fr, 2017-05-15:
After clicking on "mag calibration", this is what I see in the log panel of the configurator (1.7.0) :
2017-05-15 @ 22:49:19 -- Magnetometer calibration started
2017-05-15 @ 22:49:49 -- Magnetometer calibration finished

digitalentity, 2017-05-16:
There is no signalling from FC to Configurator about compass calibration status. Configurator assumes 30s duration (default) for the moment.

closed 2017-05-16 by digitalentity