iNavFlight/inav #1684: VFAS is only sent in single cell-mode

MarenB, 2017-05-15:
Hi, 

I think I found a bug in iNav 1.7. 
FC: SP Racing F3
RX: X4R-SB
Telemetry on UART3.

The telemetry signal VFAS (total battery voltage) is not sent, when I set "frsky_vfas_cell_voltage = OFF". I once received it for a second after the board rebooted, but then it stopped. All other signals are received fine.
If I set it to "frsky_vfas_cell_voltage = ON", I receive single cell voltage continously, no problem this way.

DzikuVx, 2017-06-13:
Hi, Unfortunately, I can not confirm the issues you describe. All my setups with SmartPort receives VFAS as expected without any problems. I suppose problem lays somewhere else.
If you find any new data on the topic, please provide it and feel free to reopen this issue.
Thanks,

closed 2017-06-13 by DzikuVx

MarenB, 2017-06-17:
I will try to reproduce and take a video of the phenomenon. I am pretty sure it worked w/o issues in 1.6.1 and it does work for me in some multirotors with CF 2.0.2 and ßF 3.1.7.
It only occured in that one airplane in which it stopped woking well after I updated to iNAV 1.7.