iNavFlight/inav #1685: How does the program achieve alt hold?

liutairan, 2017-05-15:
I just have a question that how does the program achieve alt hold with varying battery voltage? For example, when the battery is fully charged, the throttle may need to be set to 1400, but after the voltage decreases, the throttle may need to be set to 1600 or 1700 to hold altitude. The nav_mc_hover_thr variable is preset, which means at some point it may able to hold altitude, but at other points it may not able to hold altitude. So how does the program achieve this for the whole process?

DzikuVx, 2017-05-15:
thanks to GPS and barometer. And sometimes sonar. It knows altitude reported by those sensors, so can compute vertical speed. If speed should be 0 (AltHold) but reported speed is different than 0, it increases or decreases throttle. The same mechanism as for stabilization. It is also archived by PID controller (or rathrer one P and one PID)

closed 2017-05-15 by DzikuVx

liutairan, 2017-05-15:
@DzikuVx Thanks

liutairan, 2017-05-16:
@DzikuVx But what confuses me is: 
1. When I use manual mode to tune the nav_mc_hover_thr, I need to set the throttle to around 1700 to make it hover, but if I set nav_mc_hover_thr to 1700, it will jump once I turn on ALTHOLD. If I decrease nav_mc_hover_thr to around 1510, it will not jump so fast any more, but comes to the second case.
2. With the same nav_mc_hover_thr value, after I enable ALTHOLD, sometimes it will jump, sometimes it will fall. For example, maybe this time it will jump, then next time it will fall. Sometimes the propellers almost stop once I turn on ALTHOLD.

So what is going on here?

I am using SP RACING F3 board, with iNav 1.6.0, all sensors seem working well.
nav_mc_hover_thr is set to 1510, nav_use_midthr_for_althold is OFF.

digitalentity, 2017-05-16:
1. You need to set `nav_mc_hover_thr` to the value that is actually delivered to motors in hover. Stick position is NOT that value.
2. When you engage althold the quad will record current altitude and will attempt to keep it. If you were going up - the quad will reduce power (down to almost zero throttle) to stop the quad and bring it back down. If you were descending - it will increase power and gain altitude. If you activate alt hold from hovering - with properly set parameters the quad should not change throttle much.

liutairan, 2017-05-16:
@digitalentity Yes, in 1, I find the stick position is about 1700 to make it hover without ALTHOLD, then I set nav_mc_hover_thr to 1700, then save and run with ALTHOLD. Then it will jump. Later I set nav_mc_hover_thr to a smaller value and save and re-run again. 

Yes, I totally understand what you are saying and the process is exactly what I understand. That's way I felt confusing when I got it behave strangely. I will have more test to check if anything is wrong.

Thanks~

digitalentity, 2017-05-16:
No problem. I understand it may be confusing as many will expect ALTHOLD to keep throttle near hover all the time. This is the case when altitude is not changing much, but if we have non-zero climb rate AltHold will change throttle to compensate.