iNavFlight/inav #1686: Feature request: simple LED control in LED Strip tab

Ozyris, 2017-05-15:
Hello.

Is it possible to add simple ON/OFF feature on LED Strip pin to control illumination mosfets? For example when arming motors send digitalWrite High on LED Strip pin.

skaman82, 2017-05-15:
I think you can configure that already. Just select black as off color for disarmed.

Ozyris, 2017-05-15:
Is it simpler than change pin state? With this feature possible many things, not only switch lights.

krzysztofmatula, 2017-05-15:
Making binary output from led-strip is pretty easy. All you need is a module like this one: http://allegro.pl/modul-led-ws2811-rgb-ws2812-neopixel-arduino-avr-i6765766807.html

You can easily replace led with mosfet to control anything. And you can chain a number of such devices. Alternatively you can just use the ws2811 chip, but module like the mentioned one is easy to use and costs close to nothing.

Bobrcnut, 2017-05-23:
Isn't that the purpose of LEDLOW?  LEDLOW is added to the Modes page when you activate LEDstrip, and linking LEDLOW to an AUX channel allows you to turn all LEDs on or off without affecting their settings or assignments, so it somehow must be changing the state of the LED digital input line.

krzysztofmatula, 2017-05-23:
LEDLOW works well allowing to torn off the lights when activated. It basically forces black color for all elements of led strip. You still need ws2811 controller.

closed 2018-05-27 by stale[bot]