iNavFlight/inav #1693: Autolaunch fails under certain conditions in 1.7

skaman82, 2017-05-16:
Launching by throwing from the wingtip works fine but it is not suitable for my type of wing. If I throw the plane giving it a hard push from behind autolaunch does not work. 

In 1.6.1 it worked without issues.

Diff and logfiles:
`diff

[1:30] 
# version
# INAV/SPRACINGF3EVO 1.7.0 May  3 2017 / 11:13:56 (104b66c)

# resources

# mixer
mixer FLYING_WING

# servo
servo 3 1000 2000 1674 -100 -1
servo 4 1000 2000 1344 100 -1
servo 5 1000 2000 1500 -100 -1

# servo mix

# feature
feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature LED_STRIP
feature DASHBOARD
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 1 2 115200 38400 0 115200
serial 2 64 115200 38400 0 115200

# led
led 0 0,0::a:0
led 1 1,0::a:0
led 2 2,0::a:0
led 3 3,0::a:0
led 4 4,0::a:0
led 5 5,0::a:0

# color

# mode_color

# aux
aux 0 0 1 1300 2100
aux 1 2 0 900 1300
aux 2 3 2 1300 1700
aux 3 9 2 1300 1700
aux 4 8 2 1700 2100
aux 5 10 0 1700 2100
aux 6 14 10 1300 2100
aux 7 27 4 1700 2100
aux 8 28 5 1700 2100
aux 9 11 2 1700 2100
aux 10 12 4 1300 2100
aux 11 15 5 1300 2100

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6500
set acczero_x = -2
set acczero_y = 34
set acczero_z = -111
set accgain_x = 4088
set accgain_y = 4086
set accgain_z = 4033
set mag_hardware = NONE
set mag_declination = 230
set baro_hardware = BMP280
set rssi_channel = 16
set serialrx_provider = SBUS
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set align_board_roll = 50
set align_board_pitch = 10
set align_board_yaw = -900
set current_meter_scale = 293
set nav_extra_arming_safety = OFF
set nav_rth_altitude = 3500
set nav_fw_bank_angle = 35
set nav_fw_pitch2thr = 15

# profile
profile 1

set fw_p_pitch = 2
set fw_i_pitch = 16
set fw_ff_pitch = 70
set fw_p_roll = 4
set fw_i_roll = 8
set fw_ff_roll = 35
set fw_p_yaw = 0
set fw_i_yaw = 2
set fw_ff_yaw = 0
set fw_p_level = 11
set fw_i_level = 14
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 450
set roll_rate = 55
set pitch_rate = 55

#`

[LOG00004.TXT](https://github.com/iNavFlight/inav/files/1004832/LOG00004.TXT)

skaman82, 2017-05-16:
Another log with failed launchmode at the beginning
[LOG00008.TXT](https://github.com/iNavFlight/inav/files/1004844/LOG00008.TXT)

digitalentity, 2017-05-18:
You are throwing your plane to steep up. The moment you throw it the plane nose is raised up over 45 deg. This prevents launch mode from detecting a successful throw. You should be throwing your plane more or less level.

brat002, 2017-05-18:
It is one more reason to enable launch behaviour immediately after enabling
Launch mode, but not after throw. 1% people uses catapult. 99% people throw
their planes by hand. I want to 100% know if plane is ready to climb after
throw without any "smart detection", which is working from time to time.

digitalentity, 2017-05-18:
@brat002 please stop bragging about launch mode not starting motors right after enabling and chopping your fingers off. Nobody is forcing you to use launch mode - you are free to launch your plane in ANGLE mode.

If plane is thrown with more than 45deg pitch or bank angle - launch is likely to fail anyway unless the plane's motor is really powerful.

brat002, 2017-05-18:
It is not so easy to understand how much is angle - 32. or 45.8. Also gusts
of wind may do some changes in throw angle.
I am just telling you - current implementation of launch, what coming from
APM world, is far from ideal and safe.
P.S. I am not using launch mode more. Angle works great for launch.

digitalentity, 2017-05-18:
INAV's implementation of launch mode is not coming from APM - it's entirely home-brew. If you launch your plane by throwing it nose up - launch mode is not your friend. 

Launch mode puts certain limitations on how you should throw your plane to make launch safe and prevent the plane from hitting ground or people with spinning motors. 45 deg angle is a balance between likely success and likely failure of the procedure.

raul-ortega, 2017-05-18:
@digitalentity I didn't know that throwing the wing with its nose up more than 45 degrees would abort the launch. Last attempt I did was unsuccessful, and I didn't understand why, and no data was recorded in the sd card. I thought that It could be caused by some problem with electronics or servos, but now after reading this topic, I suspect that the failure was caused by the way in which I launched the wing. Please, could you take a look to this short video (only 6 seconds). I can listen how the motor goes from maximum throttle to the level I configured with the stick previous to launch. Thanks so much!

[https://www.youtube.com/watch?v=AltCE3M6uAM](https://www.youtube.com/watch?v=AltCE3M6uAM)

skaman82, 2017-05-18:
@raul-ortega at leas i can hear your motor is starting. I my attempt that never happened. Looks like your throw was to light and the wing had not enough ground clearance to stabilize

skaman82, 2017-05-18:
@digitalentity I think that the 45° limitation should go in the docs/wiki - otherwise you may get someone opening an issue on this again :) Testing later today

raul-ortega, 2017-05-19:
It would be nice to have a CLI parameter to configure that angle.

skaman82, 2017-05-19:
@digitalentity confirmed it working if throwing the plane level :)

digitalentity, 2017-05-19:
@raul-ortega @skaman82 let's make this angle limit a CLI setting.

digitalentity, 2017-05-26:
@skaman82 @raul-ortega done in #1730. Closing this one

closed 2017-05-26 by digitalentity

skaman82, 2017-05-26:
@digitalentity nice! THX