iNavFlight/inav #1696: Led strip support for naze32

bandita84, 2017-05-17:
My question would be how to solve naze32's support for led control. Fc: Naze 32 deluxe rev6, Inav 1.6.1. I looked around and saw that others had such a problem, but I could not find a solution. You might want to rewrite it in the CLI or replace it with something less important. Cleanflight handles the led, but we use iNav. There are more Naze32 cards, and they work fine, we will not replace them for this reason, unless it is a must. Thank you for your help

tonyintn, 2017-05-17:
The problem is the age of the Naze 32 and due to the limitation of available space and many different configuration your only option in obtaining that LED function back is to rebuild your own custom version.

bandita84, 2017-05-17:
Thanks for the quick response. Unfortunately, the dropbox file is not worth it. I'll do it for myself if you give someone a policy, how editor, how can I fix it. I do not want to give anyone else a job. I was not lucky with 1.7.0, I did not join again after installing it, I had to reinstall it with bootloader. Maybe if somebody has 1.6.1 in which the led-strip works, I'm grateful. But I really care how to build a version of my own. Thank all!

tonyintn, 2017-05-17:
You can get all the information you need using the search feature thats how we all did it just take a bit of reading all info is under docs/development.

https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md
https://www.youtube.com/watch?v=n3Z1fOQJAg8&feature=youtu.be

Documents version links has changed of course because of newer versions but the steps are the same just links are different, this is information you should be able to figure out yourself just by reading.

Hope this helps

DzikuVx, 2017-05-17:
We already had this discussion multiple times. Naze just do not have enought flash memory. Period. Like @tonyintn wrote, you can built your own version and disable things that you do not need.
Closing

closed 2017-05-17 by DzikuVx

bandita84, 2017-05-17:
Ok.  I think I change the onboard flash chip on Naze:
https://oscarliang.com/add-memory-chip-naze32-acro-flash-data/
@tonyintn can u the other download channel for your naze version? The f**** dropbox not allowed the download, send a error message: Error (429)
This account's links are generating too much traffic and have been temporarily disabled! Thanks all, Regards!

tonyintn, 2017-05-17:
look in the same post I provided the zip file in there as well, I have lots of people downloading files from me and drop box frowns on that.

tonyintn, 2017-05-17:
[inav_1.6.1_NAZE.zip](https://github.com/iNavFlight/inav/files/1009022/inav_1.6.1_NAZE.zip)

bandita84, 2017-05-17:
Thank You very much!!I ordered the memory chips when they arrived and the exchange was successful! Then I think it will work all at once and the naze32 will continue to live! best of luck!

DzikuVx, 2017-05-18:
Replacement of that memory chip will not solve the problem. This is
different flash memory. CPU internal flash is not big enough

bandita84, 2017-05-18:
Thanks for the information :( CPU is no longer worth replacing. Best Regards!

DzikuVx, 2017-05-18:
As a matter of fact, I strongly suggest to upgrade to a F3 based flight
controller.

bandita84, 2017-05-18:
I'm already on a SP Racing F3, I wonder what features will be added in the future. Best Regards!

DzikuVx, 2017-05-18:
I would rather ask what features will be removed from F1 to fit other
features...

bandita84, 2017-05-18:
You think well, things are going in the right direction. The 1.6.1 has been flying very well and steadily. He came home smoothly, circled evenly, in short, good. And indeed stability is the most important thing. Of course it's a lot of work hours... Then I'll try to honor some dollars. Also, thank you for your work. If I can help you, please. Best regards from Hungary.

top7chip, 2017-08-18:
Hello!!
Please,  I need help.
I have Naze32  rev6 and I migrate from Cleanflight but my led strip now dont work.
I try to build flash the last version of INAV for Naze with uncomment led strip line and undef blackbox  but during the compilation appears one  error in this file "light_ws2811strip_stdperiph.c"
the compiler say that "dmaRef = WS2811_DMA_STREAM;" is not declared.
Can anybody help me?
Thank you for your help

digitalentity, 2017-08-18:
LED_STRIP is not supported on F1 targets. Also, F1 targets won't be supported by new releases at all.

bandita84, 2017-09-07:
top7chip if reads the topic, you find the solution. "tonyintn" was generous and published his own custom edited inav version 1.6.1 where rgb led works. Download the link, if it does not go, I'll try to send it the dump somehow.

jorgebdc, 2019-11-24:
I see that in theory, led strip should be possible in naze32:
    https://www.youtube.com/watch?v=nt56WKJF5z0
    https://www.youtube.com/watch?v=0E_4K-H16Jk
So, I have tried to enable that feature (led strip) in iNav 1.7.3 by creating a custom firmware simply by uncommenting the following lines in /src/main/target/NAZE/target.h:
#define LED_STRIP
#define WS2811_PIN                                              PA6
#define WS2811_DMA_TC_FLAG                            DMA1_FLAG_TC6
#define WS2811_DMA_HANDLER_IDENTIFIER       DMA1_CH6_HANDLER
and adding this new one:
#define WS2811_DMA_STREAM                             DMA1_FLAG_TC6
Building the new custom firmware, throws no errors.
Uploading the new custom firmware to NAZE32, works.
But when activating (or switching on) the LED STRIP feature (in Configuration section) and saving changes and rebooting, NAZE32 crashes (I can't make a serial connection from the PC Configurator) and I have to upload a firmware without LED STRIP feature to not having problems. 
I disabled other features to free more memory, just in case it is a problem of memory, but the problem persists.
What could be the problem?
Thank you in advance.
 [image]