iNavFlight/inav #1697: Switch to MWOSD's character map

flounderscore, 2017-05-17:
I was planning to fix the max7645_symbols.h (and update osd.c accordingly).

In that context I'd like to propose one of the following:

1. Keep using beta flight's character map (limited number symbols, because large fraction is used by betaflight logo)
2. Switch to MWOSD's character map (easiest solution, most/all? symbols already present, no space to implement a iNAV logo at a future point)
3. Make a custom iNAV character map (larger effort, needs to be maintained, would allow to implement a small-ish logo and still support all the bells-and-whistles)

What are your opinions on that?

skaman82, 2017-05-17:
I like 2. the most. Never liked the betaflight logo taking so many characters for an useless splashscreen

digitalentity, 2017-05-18:
I like the option 2 as well!

flounderscore, 2017-05-19:
Ok, 2 it is!

moggiex, 2017-06-02:
Daft question, could you not make the word "iNav" out of existing characters and display that at the beginning instead? Just a thought :)

fiam, 2017-09-25:
This seems to have been superseded by several other OSD enhancements and the new tools we have for maintaining the character set.

closed 2017-09-25 by fiam