iNavFlight/inav #1704: mode GCS NAV,it will  fly away

lit1088, 2017-05-20:
I use f3 in  INAV 1.7.0, I can  poshold, but when i active GCS NAV,  it will fly away. in the GCS NAV mode ,i have active  ANGLE, NAV ALTHOLD ,NAV POSHOLD, GCS NAV  at the same time  with my phone  in ez-gui ,what can i do now?

giacomo892, 2017-05-20:
Does your phone send to the drone a correct position to follow?

lit1088, 2017-05-22:
How do I know if the correct data is sent?  i have checked my phone gps number  is  12

digitalentity, 2017-05-22:
NAV GCS merely allows your phone to control position of the quad. It's most likely the Ez-GUI issue, not INAV's

giacomo892, 2017-05-22:
@lit1088 from ezGUI map

tonyintn, 2017-05-23:
Not sure if it makes a difference but I do not believe you need to have ANGLE active.

digitalentity, 2017-09-08:
INAV doesn't validate what data is coming in from GCS. If GCS sends "fly to North Pole" - INAV will happily execute.

closed 2017-09-08 by digitalentity