iNavFlight/inav #1705: flaperon_throw_inverted missing in 1.7.0

Schalonsus, 2017-05-20:
The parameter flaperon_throw_inverted is missing in iNav version 1.7.0

Schalonsus, 2017-05-20:
ok, might be this:

"Flaperon support moved to mixer code. Can be configured for custom airplane mix as separate flaps now"

But how do i configure it now?

oleost, 2017-05-20:
I would assume adding an mix to the servos you want with flaps as input.

https://github.com/iNavFlight/inav/blob/master/docs/Mixer.md

Schalonsus, 2017-05-20:
yes ok, will try this

oleost, 2017-05-20:
@Schalonsus Could you update the wiki as soon as you understand how it works?

https://github.com/iNavFlight/inav/wiki/Flight-modes
https://github.com/iNavFlight/inav/wiki/Modes#flaperon

Schalonsus, 2017-05-21:
Don't know what i should add there.
I think everything is written down what you need to know for making a flaps mix.

Just make an additional smix to the servo you want to deflect with input 14. Then control if working right direction, if not invert the first 100/-100... done

oleost, 2017-05-21:
Ok, closing then.

closed 2017-05-21 by oleost

digitalentity, 2017-05-22:
@oleost @Schalonsus yes, flaperons are now part of smix.