iNavFlight/inav #1707: Passthru controls and Angle Mode Controls not meshing...

Rups63, 2017-05-21:
I have iNav 1.7.0 installed on an Omnibus F3 clone.  I installed the board in an S800 Skyshadow wing.  I orientated the USB plug to the front of the plane, and so set the yaw on the configuration tab to -90 degrees.  To make the sticks control the elevons correctly, I had to use the roll input for the pitch control and the pitch input for the roll control.  But, in Angle mode, the S800 is acting like the pitch plane is the roll plane and the roll plane is the pitch plane - how do I correct this?

On my transmitter, do I need to keep the pitch input on the pitch control and the roll input on the roll control?  When I had it set up like this, I couldn't find a combination of servo reversals that would make the control inputs react correctly.

Help?

oleost, 2017-05-21:
This is user support, and you already made an post in the forum. Closing this.

closed 2017-05-21 by oleost

digitalentity, 2017-05-22:
Most likely you got your servos connections messed up. Forget about mixes, trims and expos on your TX. 

1. Make sure model orientation in Configurator follows actual craft orientation
2. Make sure your TX endpoints are set up correctly. Roll stick left = 1000, Pitch stick down = 1000, throttle stick down = 1000
3. Make sure servos are responding correctly to **sticks** in `PASSTHROUGH` mode
4. Make sure servos are responding correctly to **sticks** in `ANGLE` and `ACRO` modes
5. Make sure servos are responding correctly to **plane being non-level** in `ANGLE` mode
6. Fly.

Rups63, 2017-05-22:
My issue ended up being that I was trying to reverse servo direction in the servo tab for Servo 2 and Servo 3, per the Wiki. I needed to actual work with Servo 3 and Servo 4 in the servo tab. Now it's sorted out and everything is working great!

digitalentity, 2017-05-22:
@Rups63 servo mapping to pwm outputs is dependant on boards and airframe type - it might be different for different setups. What wiki page are you referring to? If there's a bug in the wiki - let's fix it!

oleost, 2017-05-22:
I don't believe it was an bug, but I noticed it explained some outputs assuming it was an airplane mixer without notifying it.

So I removed it now.

digitalentity, 2017-05-22:
@oleost thanks.

Rups63, 2017-05-22:
Thanks gentlemen!  I was about to respond, but you've already got it sorted.  Awesome work!