iNavFlight/inav #1708: navPos[2] and navVel[2] going crazy in ANGLE mode at significant tilt

miradarya, 2017-05-21:
I believe I've noticed this earlier, but I recently reproduced it in 1.7 and recorded the event ([Diff](https://drive.google.com/open?id=0B40_6dQZRxYiRTFzdGhhTllXZDA), [Backbox log](https://drive.google.com/open?id=0B40_6dQZRxYiMmdlSm8wRTNQYjA), issue occurs in flight 3/3 at around 42 seconds, [Taranis Log](https://drive.google.com/open?id=0B40_6dQZRxYiMlNYQjBWX050UVU), issue starts at 11:27:28).

When flying in NAV ALTHOLD + ANGLE mode, navPos[2] goes crazy when accelerating e.g. with 40° pitch attitude after some seconds. The position estimator seems to think the multirotor is going down (navPos[2] decreases, navVel[2] goes negative) which in fact does not correspond to reality. ALTHOLD therefore increases throttle up to max, which leads to an uncommanded climb.

[image]

As you can see, baro altitude and navPos[2] diverge during the maneuver, with baro altitude actually corresponding well to what was really happening, i.e. the quadcopter was climbing from 10m to almost 30m during the event. The Taranis log shows that GPS altitude looks similar to baro altitude. So, the false decend the position estimator calculates can neither come from the baro nor from GPS.

I am somewhat familiar with coding and did take a look at the position estimator code, but I don't fully get the math behind it and therefore could not really figure out what was happening and where the false decend was coming from.

digitalentity, 2017-05-22:
Attached log doesn't have the Log 3/3

miradarya, 2017-05-22:
Oops, sorry about that, took the completely wrong file... This one is correct: [Log](https://drive.google.com/open?id=0B40_6dQZRxYiOWhBNHBTVXVpNjA)

digitalentity, 2017-05-22:
Known issue - hardware vibration problem. Vibration beyond certain level causes accelerometer to go crazy and report acceleration that has nothing to do with reality. Usually happens when throttle goes above certain level:

[image]

Solution: soft-mount your FC.

miradarya, 2017-05-22:
Ok, thanks, that makes sense! I didn't think about looking at the acc traces this time because they always looked pretty clean in other situations. I'll close this then and fix it by soft-mounting.

closed 2017-05-22 by miradarya