iNavFlight/inav #1710: OSD Speed Icon Incorrect & Blinking OSD

moggiex, 2017-05-21:
Howdy, 

Used iNav version 1.7.0 this morning for a new wing. A fantastic maiden, however two things I noticed were:

#1 The speed icon is just two arrows
#2 The OSD "blinks" a lot (is that normal?)

Obviously if these are know issues, close the thread.

The video of the flight is below. See 2:59 for the OSD blinking and also the same time stamp for the GPS speed indicator on the left showing 26 and then two down arrows.

The same down arrows appear on the summary screen at 08:32 in the video

[https://www.youtube.com/watch?v=uwyQa6C7OXQ](https://www.youtube.com/watch?v=uwyQa6C7OXQ)

Matt 

PS. Yes I know the current sensor is not calibrated, 71A LOL!

moggiex, 2017-05-21:
Whoopsy, #1 is a duplicate of this one https://github.com/iNavFlight/inav/issues/1709 Matt

Redshifft, 2017-05-21:
I would say you are fortunate anything works there at all with that installation and connectors jammed into the foam etc :D 
It's doubtful your problems are in the least inav related - test the osd on the bench with motor running etc.

flounderscore, 2017-05-22:
Should have been fixed in #1688. Let me know if there are still issues.

moggiex, 2017-05-22:
@flounderscore, yay I hope that works 👍 Does that also solve the flickering?

Note: I am using a common ground for the vTx & Camera

@Redshifft if you look closely, that is quite a neat installation. The top tray needs to fold upwards to allow the installation of the battery under it. Granted the pins to hold the tray still is not idea, I'll work out a more permanent solution today.

Matt

Redshifft, 2017-05-22:
@moggiex  People often don't pick up EMI until it manifests in the video but it's a battle with setups in confined spaces and indeed some boards are just noisy (bad designs and cheap components) the result is unwanted glitches to the boards sensors and even flight controls Personally I would not use boards with onboard current sensors, just to keep heavy conductors away from those highly sensitive sensors.

After you keep cables and wiring as short as possible ensure you turn off active motor braking for planes  if you can't find a good old simonk ESC with a capacitor on the back you may need to add one ;) sadly after all that still some cameras don't play nicely with OSD's although i doubt the current Runcam cams will have this problem.

Granted this installation is a great leap forward from your original "on wing" installations ...Lol

moggiex, 2017-05-22:
@Redshifft the ability to keep an eye on the current is an important one for me, then I know what kind of flight efficiency we have at that current point in time. 

With regards to the flickering on the OSD, is that related to the wiring? 

I know it looks a lot in the video, the servo wires are using extensions, however the power lines are as short as I can make them and it's as tight as I can make it in that model considering the space given and the requirement to lift the tray to fit in the battery.

moggiex, 2017-05-22:
And YES, better than the motor box originally used. But then again the spare Teksumo was all I was willing to sacrifice if it went in the ground ;)

flounderscore, 2017-05-22:
The MAX7456 is known to have issues with certain (usually >600TVL) cameras (check MWOSD and MinimOSD for references). I think the issue is that it doesn’t properly detect the sync signal in the video stream. As already mentioned, this can also be caused by parasitic spikes (EMI).

MWOSD has an option called “Fast Pixel” that supposedly improves higher resolution cameras, not sure whether that is used in the betaflight/inav OSD implementation.

Redshifft, 2017-05-22:
@moggiex Best test under load on the bench, you can loop out the OSD and narrow it down etc... It can be so many things, remember every wire can act as a aerial transmitting - receiving noise

Talking of servos, we have a new FC board under development with proper servo style 2.54mm connectors and heavy duty interconnects on the power and return output rails to be able to connect servos directly to the FC in iNAV which should tidy up installations by not requiring power and ground to be connected directly to servo BEC's + signal and earth back to the FC.

moggiex, 2017-05-23:
@flounderscore it's a RunCam Mini Swift, so at 600TVL

@Redshifft I did a wiring overview video on the C1 install last night, it's [here ](https://www.youtube.com/watch?v=J_DWaMh-Y_I). I'll bring the wires up over the top, but with that amount of space any changes will be limited. 

Also the FPV image was steady throughout the flight, just the OSD flicking. So still not sure on the cause, other than "tidy the wiring up".

Matt

moggiex, 2017-05-23:
@Redshifft, going off topic to reply to your comment below:

_Talking of servos, we have a new FC board under development with proper servo style 2.54mm connectors and heavy duty interconnects on the power and return output rails to be able to connect servos directly to the FC in iNAV which should tidy up installations by not requiring power and ground to be connected directly to servo BEC's + signal and earth back to the FC._

A couple of things to consider to help new users & also things I'd want to see on the board:

- At least a F4 chip because the faster it is.... (marketing speil can be easily applied, esp to a F7 chip)
- OSD. Must have an OSD. The biggest weakness of the Pixhawk is not having one integrated
- 3 UARTS that can be used besides the USB port
- Wiring/plugs included at a minimum. 
- If you want pukka quality, stick it in a box like the vector & simplify the wiring by providing connectors that make it "Matt Proof" to connect it
- Must have RSSI in on a pin not a minute solder pad which is a barsteward to solder to
- At least 6 PWM channels
- Think how a GPS unit will be connected to it. [These ones](https://www.banggood.com/Ublox-NEO-M8N-Flight-Controller-GPS-with-Protective-Shell-for-PIX-PX4-Pixhawk-p-1005394.html?rmmds=myorder) have different plugs  as they're meant for a Pixhawk, but work outstandingly well
- SD card slot, with a card included (as the jolly with the Naze32 board someone shouted at me for not having blackbox logging enabled)
- A push button boot button for ease of use
- A simple method to independently power the servo +5V rail from a separate BEC, "IF" desired.

Hope that helps,

Matt

closed 2017-09-08 by digitalentity