iNavFlight/inav #1717: Fixed-wing two motor issue

cabinw, 2017-05-23:
I've set up a plane with twin motor. All using BLS escs.
I chose multishot and esc refresh rate 2k.

This setting caused a problem that one of the esc which was in the pin 2 cannot be detected by BLHeliSuite. The esc will work just fine at low throttle. When the throttle is above almost 50% the motor makes strange noise.

This cause three escs cannot be detected. All of them work fine with Betaflight before I put them on iNav.

Really don't know why. Should I plug all escs on pin 1?

digitalentity, 2017-05-23:
Wires to ESCs are too long. Twist signal wires from FC to ESCs, connect signal ground, solder a 1000uF capacitor to power lines near ESC.

For further support please use RCGroups forums.

closed 2017-05-23 by digitalentity

cabinw, 2017-05-23:
@digitalentity Thanks！I'll use RCGroups next time