iNavFlight/inav #1720: Save home position until battery disconnect

Schalonsus, 2017-05-23:
If we disarm accidentally in-flight, home position is lost.
Would be better if home position is saved until the battery is disconnect/cycled.

stronnag, 2017-05-23:
Not for my use cases ....

Feldsalat, 2017-05-23:
Maybe  selectable by user? FIRST_ARM or EACH_ARM...  would be great!

stronnag, 2017-05-23:
As we already have that reset for baro, and home is a 3D position, just one configurable item would probably do.

sppnk, 2017-05-24:
I agree. But for god sake, a safe defined default value :-)

iconicj, 2017-06-08:
This is already implemented in the gui, at least.

There's options for home setting with each arm or with first arm.

digitalentity, 2017-06-08:
@iconicj no, this is for altitude resetting. Home position is always reset on each arm.

marbalon, 2017-06-08:
It is really good idea I think. First is when user disarm model accidentally, and second (I hope will never happened again ;-) ) when model dissarm because of bugs in software - this happened to me once and once for my friend in the past. 

@digitalentity @martinbudden are you for or against this idea ? IMHO SAVE on first arm should be as default for planes.

digitalentity, 2017-06-08:
I agree that this is a useful feature. We need an option for that. Also, what @stronnag is suggesting is viable - reset home position along with altitude. We'll have that in 1.8

digitalentity, 2017-06-08:
And default should be "first arm" - this is safe as most people do only one flight per battery.

giacomo892, 2017-06-08:
Maybe a stick combination to save current position (before takeoff) as home could be good since I like to land my quad them move some meters away and do a new flight without power cycling and with the new "FIRST ARM" option will make the quad to always return to old home which can be even occupied at that moment. 

For instance I often take off from a parking place and imagine if the spot gets occupied and failsafe kicks in :) ---> landing on car.

digitalentity, 2017-06-08:
We already have a "HOME RESET" flight mode that will move RTH position to current position of the machine once activated.

giacomo892, 2017-06-08:
That's ok. But needs a switch on the radio which are beginning to run out on my 8ch PPM :P

flounderscore, 2017-06-08:
> I also have idea to implement two meanings for HOME RESET switch: second
> will not reset HOME POS but only reset failsafe landing position. So OSD
> information, and RTL will works witch HOME POS but failsafe procedure will
> land in new position. In 99% when I fly on my spots I'm not in the middle
> of huge field where model can land safely making circles. But always easy
> to find good place to land in case of emergency landing even if this is
> 1-2km away from start point.

@marbalon: How are you going to determine the elevation of the new home
position in that case?

marbalon, 2017-06-08:
@flounderscore It is not possible (without sonar ;-) ) to know new landing altitude, but always safer land far away from trees, roads etc.

iconicj, 2017-06-08:
I just did 3 flights per battery, landing to switch Gopro modes

Happy Connecting. Sent from my Sprint Samsung Galaxy S® 5

krzysztofmatula, 2017-06-09:
> two meanings for HOME RESET switch: second will not reset HOME POS but only reset failsafe landing position.
Looks interesting, but... will you *always* use it? I bet if something bad happens this will be the case that emergency landing point was not set.
What about being able to provide a few emergency landing positions as GPS coordinates before start? That would work when flying on known spots. Not sure if it makes sense...?

digitalentity, 2017-06-09:
@marbalon, @krzysztofmatula - see https://github.com/iNavFlight/inav/issues/65