iNavFlight/inav #1722: arming problem when telemetry enabled with naze32 rev 6

thewolfbird, 2017-05-25:
Hi
I decided to update my taranis firmware to 2.1 to make use of the new telemetry features, as  2.0 could not get any gps telemetry. I connected telemetry wire from naze32 telemetry port to D4r-II receiver and enabled telemetry from configuration menu. Once the board was armed all sensors were discovered and it was working fine for approximately 5 or so minutes, so I de armed.   I decided to try and rearm just  to test telemetry again, this time it would not re arm, switches were not in any nav modes, plane level and good satellite count etc. I had a look in modes tab and noticed that the arming - button was not tabing across properly onto set aux channel bandwidth.  Then inav  just locked up and I couldn't access any other tabs. Maybe naze32  cant handle the telemetry, so I restart inav and turned off telemetry from configurator and the  board now  arms within the set aux channel as normal.

thewolfbird, 2017-05-25:
sorry my mistake usb shared with uart telemetry port

digitalentity, 2017-05-26:
@thewolfbird if the problem is resolved - please close the issue.

DzikuVx, 2017-06-14:
No response, so I suppose problem solved. Closing

closed 2017-06-14 by DzikuVx