iNavFlight/inav #1723: Distance to home froze after 71min38s

marbalon, 2017-05-25:
Hi,
Recently one on my friends find new quite serious problem - part of GPS functionality stop working after 71minutes, but GPS fix, satelites, and GPS coordinates works. I can catch this on my desk without any problem using GPS simulator on my desk.

Info about setup:
- board (in development) F3 + MPU600 + MS5611 - all over SPI
- GPS NMEA G9 (GPS + GLONAS)

The same problem on 1.6.1 and on 1.7.

You can see bug in this video on time 71min38s
https://www.youtube.com/watch?v=iha6knYtMWc

When I try too catch this problem I found this function always returns the same value when problem occurs:
`posControl.homeDistance = calculateDistanceToDestination(&posControl.homePosition.pos);`

wiesiekkr, 2017-05-25:
...and BB flight record:
[https://drive.google.com/open?id=0B3GW08HpYYLJU1J0S2RBR1BaUWM](BB)

digitalentity, 2017-05-26:
@marbalon I believe you have the resources to reproduce this issue on the bench. Can you please test the https://github.com/iNavFlight/inav/pull/1728 ?

marbalon, 2017-05-27:
@digitalentity Thanks! That was the problem, more than 80 minutes on my desk and still working ;-)

digitalentity, 2017-05-27:
@marbalon thanks for testing. Merging to 1.7.1

closed 2017-05-27 by digitalentity

marbalon, 2017-05-27:
Great!