iNavFlight/inav #1724: Unable to switch ARM with INAV 1.7.0

kahira, 2017-05-25:
With a REVO+ GPS, unable to switch ARM ( GPS module enabled or not) in a fixed wings (glider)  customized  setup.
No problem whatsoever  to ARM / DISARM via sticks combos
Consistent and easy reproducible error.
Otherwise, no further disturbing dysfunctionalities with the REVO setup/operation !

digitalentity, 2017-05-25:
Config dump would help

kahira, 2017-05-25:
Dump + diff all  is attached (operational setup , with stick arming ! )
[diff all mai 25.txt](https://github.com/iNavFlight/inav/files/1029164/diff.all.mai.25.txt)
[cli dump ref mai 25.txt](https://github.com/iNavFlight/inav/files/1029165/cli.dump.ref.mai.25.txt)

digitalentity, 2017-05-25:
In your dump you don't have ARM flight mode defined.

kahira, 2017-05-25:
what about   "set fixed_wing_auto_arm = ON"  

;-)

digitalentity, 2017-05-25:
Name of the issue is confusing then - it's talking about "switch ARM".

digitalentity, 2017-05-25:
`fixed_wing_auto_arm` setting will arm the plane when you increase the throttle. It doesn't replace stick arming.

Please confirm the following:
1. When you raise the throttle plane doesn't arm
2. At the same time you are able to arm by sticks

kahira, 2017-05-25:
I stated explicitly  that the dump and the diff all refer to my operational/ working setup.
So the title of the issue is still  very much appropriate
Just load the *diff all" less the automatic arming cli command ...and try to assign a switch for arming purpose
you will immediately recognize that switch arming doesn't  work  with 1.7.0.   (as it didn't earlier in jan/feb) 

If this is too much to do  just say so and i send you  a non workable setup

digitalentity, 2017-05-25:
Attaching configuration that is not relevant to a bug is ingerently confusing and leads to wasted time and effort. 

Issue confirmed.

kahira, 2017-05-25:
Please note: I have no problems ARMING / DISARMING  the craft with all  means available to do so with the exception of : Switch armin and disarming . So lets us  focus on the latter.
Here is a dump were ARMINg is assigned to aux6  ..something as said which just doesn't work

[dump of setup with disfunctional switch arming.txt](https://github.com/iNavFlight/inav/files/1029377/dump.of.setup.with.disfunctional.switch.arming.txt)

digitalentity, 2017-05-25:
Bug found, fix pending.

kahira, 2017-05-25:
The new title is COMPLETELY WRONG . There is no problem whatsoever with auto ARM or even more so with using mode_range_logic_operator=AND
## What doesn't work  is SWITCH ARMING/DISARMING**
Please read my messages and do not distort the issue at hand !

digitalentity, 2017-05-25:
The new title described the cause for the bug. You issue is caused by `mode_range_logic_operator=AND` - you can easily verify it by changing it to `OR`.

Now if you will allow me to get back to the coding (instead of writing here) I'll fix the issue for you - and that would be less than 3 hours from bug reporting to bug fixing.

kahira, 2017-05-25:
Note : I reversed the mode_range_logic_operator from AND to OR and... ARM Switching still doesn't work !

digitalentity, 2017-05-25:
For the record - it does ARM perfectly.

The dump from https://github.com/iNavFlight/inav/issues/1724#issuecomment-304032807 with following changes to match my TX settings:
```
map TAER1234
set mode_range_logic_operator = OR
```

digitalentity, 2017-05-25:
Despite some time wasted on needless argument - issue is fixed in under 3 hours by #1725

closed 2017-05-25 by digitalentity

kahira, 2017-05-25:
1)      This not the issue !

2)      You try to find some problem where there aren’t any 

3)      Later just to cover up that you didn’t spot that  auto ARM was enabled

kahira, 2017-05-25:
That you  criticize  the very wording which is used in the inav documentation …makes it plain !

kahira, 2017-05-25:
…don’t use words that you can’t spell properly in English.. just to cover up with  fancy words your inherently flawed logic

kahira, 2017-05-25:
I did that …and as already noted ..it doesn’t help  a bit   .

So stop shooting from the hip  ..and find something better

kahira, 2017-05-25:
Well I’ll top that …. I lost all of my time to have no problem solved in 3 hours ¨

skaman82, 2017-05-25:
@kahira wow, please calm down! This is a free project that is developed in free time and not some company software with a tech-support crew on standby. So please stop being a jerk and cut @digitalentity some slack and most importantly be polite!

kahira, 2017-05-25:
For the record also …. this DOES NOT  allow to switch arm my setup on a  Revo  / Inav 1.7.0 /  customized  Fixed wing with the setting you’ve specified below  .

The problem is elsewhere  and /or your not using the same FC with EXACTLY the same settings as I do

 

For me anyway “set mode_range_logic_operator = OR”   isn’t the solution 

 

What’s more it displays on my setup only another major   anomaly 

Indeed assigning  a switch for arming purpose  doesn’t disable the stick combo arming effect but function as if it was an “AND” condition and continue to require stick operation to effectively arm the craft .

 

The arming code seems to be truly  a convoluted mess !

 

The dump from #1724 (comment) <https://github.com/iNavFlight/inav/issues/1724#issuecomment-304032807>  with following changes to match my TX settings:

kahira, 2017-05-25:
Mr. Kravcov,

 

Did you read all of the thread .. if not please do so.

You will then  have to recognize   that right from the start Konstantin Sharlaimov  was dismissive and  peremptory.

This prompted my  reaction using the same tone has been confronted with.

So with all due respect  if you feel  lecturing  anybody  consider addressing both actors and not only me.

In any case you have broken the first golden rule which to refrain using ad hominem qualifications  

 

So learn first yourself not to be an  “ impolite  jerk “

 

Greetings / FV

skaman82, 2017-05-25:
@kahira Yes, I have read the whole discussion. I am just saying that behaving like an angry customer on a support line is not going to help you at all on github and the bureaucratic tone is also not helping at all to gain sympathy. Thats just not how it works.

digitalentity, 2017-05-26:
@kahira you came to a free project and asked for help. When I requested a dump you provided a dump from a working setup. Making a comment about that is not an excuse of providing a misleading and irrelevant information. After that you started behaving as if you purchased a product and want the issue solved by a support team and ended up calling me dismissive, peremptory, and trying to cover flawed logic by words.

If I was indeed like you call me - I'd close the issue right away with a comment about not being able to reproduce it (and that would be true btw). Please respect the time people are devoting to this **free** project. I would also expect an apology from you, but apparently that won't happen.

You issue is resolved - your dump from https://github.com/iNavFlight/inav/issues/1724#issuecomment-304032807 works with latest *development* firmware.

kahira, 2017-05-26:
Accusing me (as you do below,not commenting)  of  providing  willingly “ misleading and irrelevant information” when I supply explicitly  background information a dump + diff all  of my working environment belongs to the realm of pathology.

Your tone has been right from the start abrasive, dismissive and apodictic and needs as such to be confronted.

Ultimately , fact is that you :

1)      ignore/ dismiss  that your alleged solution to the reported issue simply doesn’t work (  its effect is  detailed and leads  to even more anomalies)  

2)      dare nonetheless  close unilaterally the issue with  even 2 successive misleading rewording of the original submitted  issue

simply disqualify you for what you are supposed to do : find real working solution for real reproducible problems.

I have  already witnessed    elsewhere this     knee jerk/Slaphappy   contribution  style of yours.

Since nothing serious  is expected to come on this basis I wish to avoid any further contacts on this matter …that is at least and explicitly with you !

 

You issue is resolved - your dump from #1724 (comment) <https://github.com/iNavFlight/inav/issues/1724#issuecomment-304032807>  works with latest development firmware.

digitalentity, 2017-05-26:
@kahira **all** configuration dumps you provided exhibit no "Unable to switch ARM" bug on latest development firmware therefore the original issue is resolved and thus closed.

You don't seem to understand that nobody here is obliged to do anything for you. If you want prompt and always polite support - go buy an EagleTree Vector, FeiyuTech Panda or other proprieary autopilot.

I don't see a point in continuing this discussion any further.

kahira, 2017-05-26:
@Kravcov 

 

Well if you did read the whole thread ..what about the pathetic   accusation thrown at me as  willingly “ providing irrelevant and misleading information” ? 

Is that what help to provide sympathy ?

Is that- among other niceties   of the same caliber-   that an  honest broker (which  you continue to  pretend to be)  can ignore ?

..and to  that you add for a good measure  that I use a  “bureaucratic “ tone   …..ach du lieber Augustin… wo haben Sie denn so  was gefunden  ;-)

Come on !

flaviosurfizera, 2018-01-23:
HI MY FRIENDS, SORRY TO CAPS LOCK,,  I AM BRAZILIAN MY ENGLISH IS BAD,
I HAVE INAV, ON COPY NAZE 32 BOARD,.
I USE THIS ORIGINAL POST  TO HABILITE "set fixed_wing_auto_arm ON" <<--- see spaces,
varius guys, talk in "no habilited auto arm",, ,in good luck, i put my "stick YAW --to rigth ...,and   habilite arm,, on stick,., i put my stick yaw to left disable arm,.

good lucky my friendss

PR #1725 (2017-05-25): Fix bug with non-working switch-ARM if using mode_range_logic_operator=AND
Fixes #1724