iNavFlight/inav #1726: Further to issue " Unable to switch ARM with INAV 1.7.0 (#1724)"

kahira, 2017-05-25:
Above issue has been hastely and unilaterally reworded to 
" Unable to switch ARM with INAV 1.7.0 when using mode_range_logic_operator=AND (#1724) "

Furthermore it has been  prematurely  closed  without checking with the submitter if the alleged cause of the error stated and its alleged correction did indeed solve the reported issue.
In fact it was peremptorily  closed   altough  the submitter notified that  reverting mode_range_logic_operator=AND to OR  clearly didn't solve the issue.
All of this is disturbing and doesn't bod well for the development of INAV (See details of issue 1724

Please advise / moderate

digitalentity, 2017-05-26:
Please provide exact steps to reproduce the issue (including what hardware was used and how firmware was configured).

kahira, 2017-05-26:
From where we stand there are several ways to address the switch arming issue in a systematic way :

 

1)You claim  that simply   reversing  “ mode_range_logic_operator”  from “and” to an “or” logic in  1.7.0 restore the proper functionality of the switch based arming of a craft. I repeatedly pointed out that this workaround  doesn't work in my environment and commented thereby also  on observed  disturbing sides effects.

If this  truly work in your environment  ,  please provide me with your firmware settings (aka dump and diff all) + confirming that you are  using  indeed  a REVO hardware . This would help me pinpointing rapidly the cause of the diverging results.

 

2)You point at a  new development release of INAV software where everything regarding ARMING /DISARMING  would be fine and dandy. Please indicate where/how to access                                                       such release of this firmware for a Revo FC.

This would help to check rapidly check if indeed the problem in my environment also is  as claimed  fully  solved .

 

3)Me providing you for 1.7.0 again but this time clear unequivocal  means to replicate simply and exactly on both sides the same environment displaying/reproducing said  switch arming dysfunctionality.

 

 

the variant 2) seems to be the “directissima” to clarify  quickly and concussively where we stand :that is , what truly  holds and what doesn't.

my  second choice  would be 1)

the least preferred  approach is 3) since it looks back ..and that anyway you have “somewhere “ found a bug which needs to be corrected 

 

What’s your take ?

digitalentity, 2017-05-26:
If you can't build the `development` firmware yourself, I suggest we wait until 1.7.1 is out (in a few days). It will include the fix for `AND` vs switch-arming bug. Then you'll be able to check if the issue is indeed resolved.

kahira, 2017-05-26:
Ok let’s wait for 1.7.1

…by the way, will this release address the RTH  issues which prohibits its use  for  fixed wing ?

flounderscore, 2017-05-26:
@kahira Could you please point me to the RTH issue. RTH works like a charm on fixed-wing for me.

kahira, 2017-05-26:
Hi there .. you seem to  have enjoyed the thread to  dare address  the bad guy  ;-)

 

Here is “l’avis de gros vent” about RTH with 1.7.0  and fixed wing(s)

 

https://github.com/iNavFlight/inav/releases

Known bugs

*	RTH descent might not work at all, be too slow or too aggressive and don't respect the landing speed on a fixed wing

For your info ,I’ve managed to embark on a Vladimir Snipe  a mini REVO + GPS + XSR ,  with total  up air weight of 250 grams  ..something completely impossible with Ardupilots available options. 

 

Reason for which I lost interest in Ardupilot  the very  platform on which you base your value added services .

 

So welcome to INAV

kahira, 2017-05-26:
This may also  interest you

https://quadmeup.com/inav-can-do-automated-landing-on-fixed-wing/

kahira, 2017-05-28:
I found the reason why (when setting  “mode_range_logic_operator “ to OR  with 1.7.0 ) our  respective environments diverge as whether  switch ARM  does or does not properly work.

There is indeed an AND logic buried  somewhere in the code which allows only  switch arming to operate if the actual throttle PWM value is at its lowest (motor not running)  .This  make sense to prevent mishaps with motorized  crafts  but none   if  the  craft has  no  motor as it is in my case.

My setup is indeed  for a  motor less thermal glider (DLG SNIPE) where the motor notion only  faintly relates  to initial external provided  kinetic energy , gravity , air movement- and resistance .

In such a setup  the  throttle stick is typically  used  (stick toward pilot) to control the amount of flap deflection which enables(among other things)  to slow down the craft.

Actually this extra   safeguard  once known  is not a problem since there are  simple workarounds to accommodate it . 

In time  however  it would be nice to have INAV  include   motor less gliders as full-fledged member of the fixed wing family  and have their  specificities accordingly taken care of.  

Finally   there is no good  reason left now  not to believe that the bug you have identified  when setting  “mode_range_logic_operator “ to  AND  will be duly  fixed  with 1.7.1. and that generally switch ARM will work as designed.

 

Please close therefore  the issue  #1726 .

 

Thank you.

digitalentity, 2017-06-02:
@kahira oh, you are trying to arm with throttle above zero! Yes, this is a necessary safeguard to prevent accidental arming with WOT. Glad that you figured it out. 

I'm closing this issue now.

closed 2017-06-02 by digitalentity