iNavFlight/inav #1727: Motors suddenly stopping in flight

pstryju, 2017-05-25:
Hi
First, I'm using Naze32 Rev6 10DOF with NZ GPS (Ublox-7), mounted on F450 with 2212 motors and SimonK ESC's, Inav 1.7 compiled without any telemetry, disabled software serials and with led strip.
Problem is that all the motors just stops for 1-2 seconds and quad recovers (loosing 30 meters of alt). Blackbox is showing 1000 for all motors and i can't see any reason why, no failsafe, no invalid rc command, no PID correction problem. Happend three times during 6 minutes flight. And today it's not a first time. Enabled: Angle, Althold, Poshold.

[inav17_dump.txt](https://github.com/iNavFlight/inav/files/1030444/inav17_dump.txt)
[blackbox_log_2017-05-25_165937.TXT](https://github.com/iNavFlight/inav/files/1030446/blackbox_log_2017-05-25_165937.TXT)

Third flight in blackbox -> 04:26.526; 05:20.168

Klapstuhl, 2017-05-26:
First of all you need to tune your PIDs.
I think your motors stop because Minimum Throttle is set to 1000 and your Motors start to spin at a higher value. Youse the Motors Tab in the configurator to find the value at wich your Motors start to spin and use that value for as Minimum Throttle.

krzysztofmatula, 2017-05-26:
Also proper esc calibration may help.

pstryju, 2017-05-28:
Esc calibrated. 1150 motors idling. And check blackbox. FC is sending 1000 not 1150.

digitalentity, 2017-05-28:
When navigation modes are enabled `rcCommand` shows firmare-generated values and we don't have information about actual pilot's input. You are probably moving your throttle stick below `min_check` - you have `MOTOR_STOP` enabled and it's the only possible case (apart from disarm) where motors could be stopped in flight.

digitalentity, 2017-05-28:
Solution - disable `MOTOR_STOP`

pstryju, 2017-05-28:
Wow. That is really important that rcCommand is let's say after navi modes. I think you're right now - throttle stick to low. Going to disable MOTOR_STOP and check.

parnz, 2017-05-28:
Exactly the same problem here just a few minute ago bottom frame broken
will try disable stopmotor when get a new frame

spracingf3

parnz, 2017-05-28:
forget to add: in Nav poshold activated

urlu75, 2017-06-02:
had the same issue when using PPM RX (X4RSB) with PPM activated (with inav 1.6, 1.7 not tested yet)
Motors stop after a few sec with no reason and could restart just after if moving throttle stick.
To bypass this issue, I had to change the frsky PPM config, channel CH1 - CH8  (1 to 16 by default).
At that time I was thinking that it's a frisky X4R issue because it only have 8ch.
Don't know if it it's related...

DzikuVx, 2017-07-04:
Sine no new facts appeared in last 4 weeks and looks like hardware  issue, I'm closing this one. If any new data appears, feel free to reopen

closed 2017-07-04 by DzikuVx