iNavFlight/inav #1731: SUMD with Receiver Graupner GR12L can't be enabled

ZerwasMOD, 2017-05-28:
Good Morning,

I am excitingly trying to setup up you software for my flying wing and are running into an issue and I am stuck now.

I want to enable SUMD and I don't get it working on iNAV.

This is my setup.

Receiver: Graupner GR12L set to SUMD at port 6
FC: Naze32 Rev5 Signal is soldered to port 4 

I tried the following setup. In the ports tab, I enable for UART2 Serial RX in the RX column. This is fine, can be saved and stays enabled.
In the configuration tab, I want to set the Receiver Mode to Serial-based receiver (...) and set the Serial Receiver Provider to SUMD
When trying to save the configuration tab, the Receiver Mode: Serial-based receiver does not stay saved and after rebooting no radio button is set. Also if it try to set feature RX_SERIAL in the CLI and save it, it doesn't stay enable.
Additionally, I set, in the Receiver tab, the Channel MAP to TAER1234 but the I don't see any of sticks bars where they are supposed to be

When I flash the FC with Betaflight and do the exactly same settings I can run SUMD there flawless.

Attached are my dumps with a freshly flashed iNAV and only tried to set the setup as explained above.

[status.txt](https://github.com/iNavFlight/inav/files/1034000/status.txt)
[dump.txt](https://github.com/iNavFlight/inav/files/1034002/dump.txt)

Thank you very much for looking into this.
Alex

digitalentity, 2017-05-28:
See https://github.com/iNavFlight/inav/wiki/Supported-boards
NAZE only supports PWM, PPM, SBUS, IBUS, SPEKTRUM receiver types and LTM/FRSKY telemetry. You have to get an F3-based board.

ZerwasMOD, 2017-05-28:
Hi,
thank you for the quick response.
This NAZE is an F3 based board, and SUMD setup works when flashed with Betaflight.

digitalentity, 2017-05-28:
There is no F3-based NAZE. This is probably a cloned SPRacingF3 board. Which firmware do you use?

digitalentity, 2017-05-28:
To verify what board you have - please provide output of `version` CLI command.

ZerwasMOD, 2017-05-28:
here it is:
# INAV/NAZE 1.7.0 May  3 2017 / 11:03:28 (104b66c)

digitalentity, 2017-05-28:
Yes, you have a NAZE board - it has F1 CPU and only 128k flash, so stock INAV firmware indeed doesn't support SUMD and HoTT on that board.

ZerwasMOD, 2017-05-28:
ok, I go for PWM until an F3 board arrives. 
Thank you for clarifying.

digitalentity, 2017-05-28:
No worries. Good luck flying! I also suggest you wait a few days until INAV 1.7.1 is out - we've fixed some serious bugs there.

digitalentity, 2017-05-28:
I'm going to close this issue now.

closed 2017-05-28 by digitalentity

Klapstuhl, 2017-05-28:
So you can use Sumd on an F1 board but you need to compile your own version of iNav.

digitalentity, 2017-05-28:
Yes, @Klapstuhl you are correct. It's possible to build a custom INAV firmware for NAZE board with SUMD + HoTT.