iNavFlight/inav #1735: Calibrate Gyros Not Documented

teckel12, 2017-05-29:
I see that you can calibrate the gyros by moving the left stick to the bottom left and the right stick to the center bottom.  But, I can find find no information in the documentation or via a Google search for what to actually do to calibrate the gyros.  There's info on the accelerometer and compass calibration, but not the gyros.

Maybe I'm missing something in the documentation, but as the documentation is (in a repo) it's impossible to find anything.  On a side note, there should be a main documentation page that organizes the different doc pages so things can be easily found.  Like a readme.md in the doc folder.  Or, some kind of index or something.  The doc file names don't help.  For example, shouldn't there be a "Modes" doc?  Where would one go to find the flight modes?  With an main documentation page, this could be organized in a way where one would have some kind of hope of finding something (unlike it is today).

giacomo892, 2017-05-29:
Gyro does not need a manual calibration.
The natural bias all gyros have is estimated at machine boot and when it is not moving.

teckel12, 2017-05-29:
If that's the case, the documentation should be changed [as this] (https://github.com/iNavFlight/inav/blob/master/docs/Controls.md) shows how you go into calibrate gyro mode (which created the confusion.

stronnag, 2017-05-29:
The documentation is correct. The stick commands cause a gyro calibration. You have to do nothing else for the calibration to complete.

giacomo892, 2017-05-29:
@teckel12  You may need to do a gyro calibration by stick if you move from your machine from a very hot place and fly right away. 
I think power cycling the machine after some minutes does a better job :)

teckel12, 2017-05-29:
So should it be steady, or bouncing around in the bed of my truck when a gyro calibration is done?  I ask because it doesn't specify, and for other calibrations all kinds of particular movements need to be done.

giacomo892, 2017-05-29:
If it is not steady It will not calibrate.
If you feel the documentation is inaccurate just provide a pull request with modifications

rb1205, 2017-05-30:
For sensor calibration instruction you should head to the [relevant page on the wiki](https://github.com/iNavFlight/inav/wiki/Sensor-calibration). As you can see gyros are not included in the calibration checklist because it's done automatically on power on. In fact, it shouldn't even called "calibration", it's a simple zeroing.

Redshifft, 2017-05-30:
We know Gyros are calibrated at battery connection, it would be nice to know how long after connection the process starts and ends. Sure we have lights on the FC to confirm system check etc but it's the age old pain of trying to hold the craft still while connecting the battery.

giacomo892, 2017-05-30:
@Redshifft You don't need to keep the aircraft still **while** connecting the battery.
Just connect the battery with calm and with some more calm put the craft on the ground.
Gyro calibration will complete and you are ready to go.

DzikuVx, 2017-05-30:
Wiki updated, I'm closing this ticket

closed 2017-05-30 by DzikuVx

Redshifft, 2017-05-30:
@giacomo892 :) doubt very much the process is intelligent. As you touched on before if you want the best from any of the sensors it is best to let them warm up then do a power cycle certainly until temp stabilized sensors come along ;)

giacomo892, 2017-05-30:
@Redshifft if you move your machine during calibration the calibration fails. Just take a look at the gui :)

Redshifft, 2017-05-30:
@giacomo892 Your only bloody right :D Calibrates/initializes after movement stops (no way to test in the GUI as it powers the board) but you can see the boards light sequence and being that movement only interests the Gyro or Acc It looks to wait until no movement is detected.

@DzikuVx  Worth adding to the Wiki "After power up ensure craft remains dead still for 5 seconds to initialize gyros"
 I never had any issues with any of this because I go to great pain to not move things while initializing but never knew anymore than must not move it through that phase I now know how to do it. or should I say what it needs to make it happy, for Inav anyway :)

DzikuVx, 2017-05-30:
@Redshifft you are free to make that change to the wiki by yourself :) All contribution is welcomed

giacomo892, 2017-05-30:
@Redshifft heheh. Pilot just need to plug the battery and place the machine at rest. Then all the magic will happen :)

Redshifft, 2017-05-30:
@DzikuVx , sorted :)

@giacomo892  Sure, the mystery was when  that process started, most Quads get put on the ground - I suspect many Aeroplanes are just held by someone ;)

teckel12, 2017-05-30:
Thanks for the doc update.  That clears things up.  Please understand that I didn't know what the answer was, or I would have updated the docs myself.  There wasn't any mention of gyro calibration, so I had no idea what we should be doing to calibrate.  Anyway, thanks for clearing this up.

Also, I may put together a doc index/table of contents as a folder filled with docs named in unhelpful ways is not a very good way of doing a manual.  The lack of search (in just the docs) is also a problem.  Sure, you can do a repo search, but that typically just returns issues and not the manual.

Thanks again!

Redshifft, 2017-05-30:
@teckel12 No problem, I also learnt today. I think we could safely say very few knew what really happens. You have to realize some of this stuff dates back to multiwii  and we have just all used the code  in it's various forms over the years and only discussed  the more interesting developments :)

Anything you can contribute to the docs will  be of great help to new users and some sort of search  would be great.