iNavFlight/inav #1740: Air Speed Sensor Wiki Page

moggiex, 2017-05-31:
Howdy,

There isn't any documentation on the use of a Air Speed sensor in the wiki (which I can find anyway) hence this request :)

Can a page be created that covers the basics, such as:

- Which air speeds sensors are known to work with iNav
- How to connect them to a FC board
- Anything else a user needs to know, such as calibration or settings applicable to them

Matt

digitalentity, 2017-05-31:
Airspeed sensors are not used for anything just yet - hence no documentation exists to avoid confusion.

oleost, 2017-05-31:
It could be beneficial with an wiki page mentioning which airspeed sensor currently works, and how to connect it on a supported board.

Also mention that airspeed is not currently not being used for anything, but in future there may come additional beneficial of having it.

This way users building an airplane could choose the correct sensor and be ready if INAV starts to use it.

parnz, 2017-05-31:
Should be removed from Configurator to avoid confusion as well?

oleost, 2017-05-31:
No why?

It is, working. But only used for logging at the moment.

moggiex, 2017-05-31:
@oleost, then I'd say myself as an user would be confused by that "It is, working. But only used for logging at the moment.".

So a wiki page saying this would be "spot on", it can be updated later that way and clears any confusion.

> "An airspeed sensor with iNav is only being used to help with data logging currently. iNav may support an airspeed sensor at a later to help with flight modes, but for now, logging only."

Sounds ok?

Matt

closed 2018-05-27 by stale[bot]