iNavFlight/inav #1742: iNav Failsafe Issue

jstremmler, 2017-05-31:
I own a copter which from now on I **only** want to operate with a 3DR-radio and the existing SPRACINGF3 FC.
How do I avoid failsafe kicking in  because I always outside valid pulse range setting because I don't have any receiver on the copter?

digitalentity, 2017-05-31:
You need to bypass this in the firmware. INAV does not support receiver-less operation.

jstremmler, 2017-05-31:
Thanks for the info!

I will do so, but will keep failsafe active for low battery and RTH action and just deactivate  valid pulse rating   initiating any failsafe and will then compile my own iNAV-RLO version. (RLO=receiver less operation....),+++

digitalentity, 2017-05-31:
If you manage to make it user-friendly and reliable - please consider submitting a PR into this repo - "RLO" would certainly be a great addition to INAV.

GTorreil, 2017-05-31:
+1 , RLO is usual in researchers' workflow.

anbello, 2017-06-01:
@jstremmler how do you send command to 3DR-radio?
3DR-radio is like a serial link, right? If yes it should be possible to send MSP command and choose MSP as RX input mode, without the need to recompile. But maybe I am missing something.

b14ckyy, 2018-02-20:
I think RLO Could be very dangerous isn't it? 
What happens if there is any GPS issue, or in worst case an accidently wrong set waypoint far away? The copter could just run away whithout having the virtual "dog leash" of the receiver that triggers RTH failsafe. In my experience 3DR telemetry and pull messages are not reliable enough on bigger ranges to use this as an failsafe indicator. So how would you detect if the copter freaks out and goes out of sight? 

I'd recommend using a receiver mod for bigger range or at least any kind of transmitter station that sends a valid signal out to the copter (without the need to control it manually) to have a physical range limitation here.

closed 2020-12-11 by DzikuVx