iNavFlight/inav #1744: Colibri Race FC

cdog4twenty, 2017-05-31:
Cannot get the acc to recognize any axis of the 6 step calibration procedure. Is this possibly a firmware bug?

cdog4twenty, 2017-05-31:
# dump

# version
# INAV/COLIBRI_RACE 1.7.0 May  3 2017 / 11:06:12 (104b66c)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature MOTOR_STOP
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
[... 398 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 15
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

cdog4twenty, 2017-05-31:
Also if I select the gyro (6000) it marks it as red. The new colibri fc has a 6000 not a 6500 which it thinks it has.........

cdog4twenty, 2017-06-01:
Im over it, back to vector for me. The calibration procedure is dumb...................

Redshifft, 2017-06-01:
Guess it just didn't happen fast enough for you :)

digitalentity, 2017-06-02:
If somebody won't read the manual to me in 5 minutes - I ain't gonna use this firmware! LOL!

closed 2017-06-02 by digitalentity

cdog4twenty, 2017-06-02:
Read the manual, calibration isnt working on colibri............. the manual sucks, and so do you for leaving the worthless reply......

cdog4twenty, 2017-06-02:
Calibrated by press acc calibrate 6 differnt times with FC in position shown and nothing ever happens with the numbers in the CLI

oleost, 2017-06-02:
I'm sorry, but do you really think anyone is going to help you with that attitude of yours?

Maybe you found a legit bug, maybe you didn't. But I'm sure I won't use my free time assisting you when your commenting the way you are.

Anyway bye, this was the first and last comment from me in this issues post.

cdog4twenty, 2017-06-02:
No you are right Im being a complete dick. I'm sorry guys.

cdog4twenty, 2017-06-02:
I actually figured it out, Im an real asshole and again, my apology. I assumed the copter remained flat and rotated like a clock, i re-read the instructions 20x instead of just looking at the picture like a baboon, and finally got it. IM AN IDIOT, A WHINY ASS PROBLEM WITH AMERICA KINDA SHIT. So really my apology please forgive.

Redshifft, 2017-06-02:
LOL, I edited that text for ACC cal because it was confusing previously,
 I added "calibrate the FC board out of your model for simplicity" There is even a link to a video :)
bottom line, top,bottom and sides down , basically if it was a box 6 sides to calibrate ;)

digitalentity, 2017-06-03:
@cdog4twenty 
Albert Einstein once said: "Anyone who has never made a mistake has never tried anything new”. We all make mistakes, but not everybody have the gut to admit them. Good to know that you figured it out eventually. 
Good luck flying man!