iNavFlight/inav #1745: inav 1.7.0 By Omnibus F4 Pro OSD wrong

hsu6328, 2017-06-01:
The latitude and longitude displayed in the OSD of Omnibus F4 Pro is wrong(INAV 1.7.0)

jstremmler, 2017-06-01:
Cant confirm that! 

Can you mail a picture from the OSD and a screenshot from the configurator GPS position?

StewLG, 2017-06-04:
There has been at least one fix applied to invalid display of lat/long since 1.7.0. Although it doesn't (currently) show up in the release notes, 1.7.1 should include this correction:

https://github.com/iNavFlight/inav/pull/1676

So, please try it again with 1.7.1. If you are still having problems, please post a screen shot, and tell us what lat/long *should* be displayed.

hsu6328, 2017-06-15:
I have tested that the problem has been fixed in 1.7.1，thanks

StewLG, 2017-06-15:
Great! Glad to hear it fixed your problem.

You should now close this issue.

digitalentity, 2017-06-15:
Since it's fixed, I'm closing this one.

closed 2017-06-15 by digitalentity

StewLG, 2017-06-15:
@digitalentity, it would be good to edit the 1.7.1 release notes and mention that this bug has been fixed.

I would submit the change myself, but I'm not sure how. The release notes don't appear to be part of the repo, and it's not a wiki page either. (If I can make/submit an edit, tell me how and I'll do it.)

digitalentity, 2017-06-16:
@StewLG  agreed, release notes edited.