iNavFlight/inav #1749: Spin like crazy and fall out of the sky -- iNav 1.7 with Ominbus F3

isilcala, 2017-06-02:
I was flying a newly built Z84 with iNav 1.7 Omnibus F3, first pack was perfect, tested Launch Mode / RTH / PH / AH / Auto Trim / AutoTune / Angle / PassThru, all were working like a charm. Watched the plane flying itself in PH mode for 15 minutes without a single glitch.

The issue happened in the second pack, I was flying in angle mode and suddenly the plane turned nose down to the ground and then spin like crazy, during that time it didn't respond to my stick inputs at all.

Luckily I was able to retrieve the plane with Blackbox log. After reading the log I can see the PID_I[pitch] first went mad(which caused the nose down I guess), a few seconds later PID_I[roll] went mad as well(caused the spin). Later even the RX input became rubbish.

I couldn't figure out what was the root cause that made the whole plane went crazy. Your help is appreciated!

Build:
Wingwing Z84, Omnibus F3, Ublox M8N, XM+ receiver on SBUS.

Dump and Blackbox log:
[Z84.zip](https://github.com/iNavFlight/inav/files/1047187/Z84.zip)

The issue happens on 10:40

Thanks!

digitalentity, 2017-06-02:
Looks like a servo failure to me. I-term windup was in the **opposite** direction than roll spin - meaning that PIFF controller was desperately trying to stop the rotation.

DzikuVx, 2017-06-02:
I also tend to agree it was a servo failure.
set servo_pwm_rate = 330
Are you sure servos were able to handle that for a longer time?

isilcala, 2017-06-02:
I was using digital metal gear servos, should be able to handle 330 refresh rate. But I'll check servo later and report back.

isilcala, 2017-06-02:
No. Confirmed it wasn't servo failure. Both servos are working fine.
Also what could be the reason of RX went crazy halfway in the spinning?

digitalentity, 2017-06-02:
I don't see RX going crazy. Halfway through the spin you activated PosHold which took over the sticks.
Blackbox is 100% clear - plane was unable to execute what firmware told it to. It could be servo failure (either final or just stuck gears), linkage failure, servo power supply failure, servo cable disconnecting etc.

Btw, what servos did you use?

moggiex, 2017-06-02:
@isilcala I have had EXACTLY the same issue with the Wing Wing Z-84.

If the chaps above are saying it's flying OK and you managed to get it to fly straight in Acro or Horizon mode, then that leaves one thing, a tip stall.

Wing Wing Z-84 **must be flow nose heavy**. If you don't they'll death spiral on you without any warning. 

I know this as I've experienced it first hand 3 times. Twice with with a vector onboard and once in a model with no kit.

So when you balance your model, put two craft pins on the front most part of the CG nodule under the wing and balance there. This will stop the wing from doing that dirty tip stall on you.

Oh and watch this video, that was in the WW Z-84 with a Vector onboard, how I saved it, I do not know! 

PS. Just maidened my WW Z-84 this morning with iNav, flawless...

Matt 

[https://www.youtube.com/watch?v=1hqyERVOnA8](https://www.youtube.com/watch?v=1hqyERVOnA8)

isilcala, 2017-06-02:
@digitalentity servos were EMAX ES08MD II

isilcala, 2017-06-02:
@moggiex gosh how did you save it!!!
Your video looks EXACTLY the same as I experienced. It even changed the direction halfway of the spinning the same!!!
I'll try to balance it per your instruction, see if that fix it. Thanks!!

moggiex, 2017-06-02:
Howdy @isilcala, I have no idea! 

And yes thats a wing wing Z-84 "death spiral". Add some weight in the nose and it won't do that again :)

Matt

Redshifft, 2017-06-02:
Yup, flat spin, not much chance of recovery with a Wing with pusher propeller because you have no air across the control surfaces so no control authority :( also a FC is unlikely to be able to recover a spin.

If you have Manual control (go passthrough) full power, down Elevator and rudder against the turn if you have a Rudder - if it won't come out cross control the Ailerons/rudder and opposite Elevator to flatten the spin it may even stop spinning with no power :)  recover if it does, if it does not it will crash so gently damage will be minimal.

As also mentioned move the CofG forward 5mm on a wing

isilcala, 2017-06-02:
I'll close this issue and try your suggestions. 

Thank you all!!

closed 2017-06-02 by isilcala

moggiex, 2017-06-02:
@isilcala This was my maiden this morning with the Z-84, I've put my settings in the video description and if you look closely, there is a load of lead under the front nose [https://www.youtube.com/watch?v=u2j4KErWH4E](https://www.youtube.com/watch?v=u2j4KErWH4E), hope that helps! Matt