iNavFlight/inav #1750: Reconsider low battery RTH please

Dutchy-79, 2017-06-02:
Hi, 

I was searching for Inav low battery RTH failsafe and ended up here https://github.com/cleanflight/cleanflight/issues/1711#issuecomment-305748321

So I asked in that thread about the future and eventually they send me here. 

I'm a APM pilot and now building a new hex with an omnibus F3 and inav. 

I would like a failsafe option for low battery like apm does. 

I already came up with a secondary solution by using telemetry and a logical switch. (cell <3.8v for 5 seconds sets aux#(failsafe channel) to the nessecary value. 

Tho I'd like this to be a future for inav. This will help people without a programmable radio or without telemetry enabled rx.

oleost, 2017-06-02:
Hi, there is already exisiting issue tracking this, its a feature that hopefully will come in INAV 1.8.

#74 #1475 

Closing this issue.

closed 2017-06-02 by oleost

Dutchy-79, 2017-06-02:
Thanks Peter, I didn't found it.