iNavFlight/inav #1753: Feature Request: OSD Stick Menu System

moggiex, 2017-06-02:
Howdy,

What are your thoughts on adding a basic OSD menu system to change settings using the ele/pitch sticks similar to what is found on a Eagle Tree Vector.

See this video at 9mins 51s to see a visual example:
[https://youtu.be/2zebDPa9eec?t=9m51s](https://youtu.be/2zebDPa9eec?t=9m51s)

The OSD menu can be activated by clicking the mode switch twice quickly, then the menu appears. Almost ALL settings can be changed in the menu system as well, including features on the OSD screen itself.

I have no idea if this is possible or not with iNav, however the ability to tweak the settings using the OSD and the Tx sticks is a simple thing to do as a pilot and comes in handy on the flight line.

Obviously I'm not suggesting a full blown menu system for every setting, but enough to get by with on the flight line with no laptop or bluetooth/phone combo.

I hope that makes sense & I can provide additional screenshots if desired.

Thoughts?

Matt

moggiex, 2017-06-02:
_"Obviously this is a wish list"_ 

The basic settings that I would personally like to see in such a menu and yes I only fly fixed wing :)

- RTH height (increment/decrement in 1M intervals, nav_rth_altitude)
- RTH cruise throttle (nav_fw_cruise_thr)
- Loiter Radius (increment/decrement in 1M intervals, nav_fw_loiter_radius)
- Bank Angle (same as the others, nav_fw_bank_angle)

Matt

t413, 2017-07-26:
+1! This has been done for betaflight and is fantastic to use. See [video demo](https://www.youtube.com/watch?v=onymhmGui78) and that [source code](https://github.com/betaflight/betaflight/tree/master/src/main/cms). Something where waypoints could be marked, modes could be entered, and settings could be changed while flying would be spectacular. Just enter loiter, RTL, or just leveling and flip a menu switch.

digitalentity, 2017-07-26:
@t413 CMS (menu system) is present and working. It lacks settings for navigation though.

DzikuVx, 2017-07-27:
@moggiex just like DigitalEntity wrote, it is there already: `Throttle stick mid, Yaw stick left, Pitch stick up, Roll stick centered`

brat002, 2017-07-27:
I strongly suggest you to see how this feature is implemented in Pitlab OSD. I guess, it is much convenient.

t413, 2017-07-27:
Great to hear this is at feature-parity with betaflight! I couldn't find any mention of the OSD menu in documentation or feature lists and haven't tested it with my Omnibus-F3 board yet.

Should another two issues be opened as feature requests for: 3-pos switch menu navigation / stick override; navigation operations in the menu system?

@brat002 nice example with the pitlab configuration. Another good comparison is the ET Vector. It uses a stick-driven OSD menu like iNav and allows for excellent in-flight configuration. Here's a great example [on youtube](https://youtu.be/8bkwFACK8jo?t=463) (links to at 7:48 in the video). 

I personally like the idea of a configurable switch-range option for 'OSD-Menu' that would immediately exit when switched off. When on it would allow for the right stick to navigate the menus as usual and hold the RC inputs for those channels in center/previous value. Next step: playing Snake while waypoint navigating.

DzikuVx, 2017-07-28:
@t413 always create new issues for new things. It greatly helps to keep track of what is going on. 
One thing = one issue

I'm closing this one, feel free to create new

closed 2017-07-28 by DzikuVx

V8Darren, 2018-07-22:
Could we please change the menu activation to a switch. You cannot enter the menu with throttle cut activated, and if you enable the the throttle the prop spins. On a switch would be safer.

leeph, 2018-09-04:
A mode-switch gesture would be good. I don't want to have to dedicate yet another switch for a rarely used function. On Vector, you can enter the OSD stick menu by 'toggling' a 2-or-3-position mode switch twice. Something like that, anyway.

wx4cb, 2018-09-14:
problem i have right now and adding on a switch toggle would be great is that i don't have a "yaw" axis configured... there's no point because it's a flying with and there is no rudder, which means I have to configure the rudder channel.