iNavFlight/inav #1756: OSD: "UK Mode" Show Speed in mph, while other values in Metric

moggiex, 2017-06-03:
Howdy,

At first glance this might sound odd! 

> Why would someone want the speed in miles per hour (mph), but the other values, such as height or distance in metres or kilometers?

Here's the thing, in the United Kingdom car speedo's are in mph, so the we understand speed in mph (and not kmph), however in schools we're taught to use measurements in metres etc...

Hence the request, is it possible to be able show mph for speed, but keep the metric values for the options?

Matt

martinbudden, 2017-06-03:
@digitalentity , while this does seem odd, it does actually make sense for people flying in the UK.

moggiex, 2017-06-04:
@martinbudden we're "special" in more ways than one :)

martinbudden, 2017-06-04:
It amuses me that all quadcopter measurements are metric, except prop sizes which are imperial. And there's even an exception to that rule - prop sizes for small brushed quads are metric.

Redshifft, 2017-06-04:
Lol, great subject. Imperial system is total crap really but I'm guessing even British kids out of school now only catch fish in pounds from water depth measured in feet ;) and If you told them how fast your car was in kph they would say..wtf is that in real speed :D 

But like that school kid, I look at the metric osd and think how fast is that really...LOL but for sure I don't need a calculator to work out how far home is in time when I have km's and meters and i'm flying in km's opposed to miles and feet.

geofrancis, 2017-06-13:
real nerds use meters per second lol

fiam, 2017-08-24:
@martinbudden Please, send this one my way too

flounderscore, 2017-08-24:
@fiam Thanks Alberto for contributing so much to iNav lately. Since you're doing a lot of OSD related work, maybe you could consider switching over to the MWOSD char map (I started working on it in https://github.com/iNavFlight/inav/pull/1796 )

fiam, 2017-09-07:
Should have been automatically fixed by #2049. Can someone with write access mark this as fixed?

digitalentity, 2017-09-08:
Fixed by #2049

commit 92dbe80a (2017-09-04): Add support for UK OSD units

UK mode shows speed in mp/h while showing the rest of the values
in metric.

Depends on https://github.com/iNavFlight/inav/pull/2048

Fixes #1756
closed 2017-09-08 by digitalentity