iNavFlight/inav #1759: I2C at AIRBOTF4 with external MPU9250/BMP280

dieideeistgut, 2017-06-03:
Hello, possibly a question/issue created by my overthinking:

I have a GY-91 external I2C module running nicely on an older NAZE Board (tested). Trying to get this thing working on the AIRBOTF4 target for some days now. Only result with the current 1.7.x versions is a non-booting FC (therefore no ´bootlog´ or ´status´ attached). I checked the GY-91s I2C addresses and they reply back with the 0x68 and 0x76 which should be fine - from my understandings. My FC uses a MPU6000 internally which maybe interferes with the 0x68 address from the MPU9250? 
Hardware-wise i've chosen a setup recommended just like this: 

[image]https://static.rcgroups.net/forums/attachments/5/6/3/4/2/8/a10059593-122-AirbotF4-Acro-iNav_ExtBaro%26Compass_01.jpg)

Now my questions for you experts would be:

1) are there any pullups needed even though the GY-91 has a 5V level converter running?
2) are there ways to give the external MPU priority over the internal one? 
3) or ist he GY-91 a dead-end anyways? 

Thanks so much (ツ)

Key0nee, 2017-06-05:
I had a similar problem trying to access the HMC5883 in my GPS puck.... added 4.7K pull ups to scl/sda and iNav recognized it... still not completely tested but at least iNav is seeing the external Mag now

dieideeistgut, 2017-06-05:
Thanks! Will try. Managed to get the internal BMP280 running perfectly now. Mag is not seen still - working on it.  Will reply. (ツ)

GTorreil, 2017-06-06:
Hi, [slightly off-topic] I used to have the same BEC as you picture in your schematic. There is a problem with it... Being an adjustable power supply module, it does not have any protection on the output in case voltage climbs too high. You should definitly fix this if you are using braking ESCs : it ruined all my onboard electronics because of a 22V spike reaching the 5V circuit! Ruined the FC, GPS and sensors onboard...

GTorreil, 2017-06-06:
or you can add a 5.6V Zener diode which might be enough

dieideeistgut, 2017-06-06:
AD0 with 3.3V will kick the MPU9250 on the GY-91 in a second state - the deviceIDs change to 0x69 for the MPU9250 and to 0x77 for the BMP280. Thats how i got the baro working now. Stuck on accessing the AK8963 baro - do i need to init the MPU9250 via MPU6500 to access the AK or can this be bypassed somehow?  Sorry for all these questions that may seem logic to anyone else here.

DzikuVx, 2017-06-28:
Since we do not have any new action here, I'm clocing the ticket. In case anything new appears, feel free to reopen

closed 2017-06-28 by DzikuVx

3sr3v3r, 2017-07-31:
Good one!
I've been trying to get the Reyax RY836AI to work and have some struggles too.
Got it working with the F3Discovery board.

Had to set the MP9250 to bypass mode to access the AK8963.
Added these lines to the AK8963 driver.
Wasn't able to get it working on a PIKOBLX and Omnibus but thanks to you remark I now realize that the MPU6000 has the same whoami address!! thanks for that one.

        //Reset MPU9250 registers
        i2cWrite(MAG_I2C_INSTANCE,MPU9250_ADDRESS, 0x6B, 0x80);
        //Set bypass mode for AK8963
        i2cWrite(MAG_I2C_INSTANCE,MPU9250_ADDRESS, 0x37, 0x02);

The  whole description about setting the MPU9250 to master on the I2C bus is applicable when accessing the MPU9250 through SPI. Not brilliantly documented and quite confusing.

tnako, 2018-04-02:
Hi,
Im also want to use compass from mpu9250 and connected via i2c
Based on @3sr3v3r idea, here is patch - https://github.com/tnako/inav/commit/a9758cb7e8f3626d63c99aec7fa01273009061cd
All bypass stuff goes to mpu9250, so in configurator need to set "mpu9250"
And target.h need #define USE_MAG_MPU9250

Currently testing with matek f405

ninja-zx11, 2018-08-07:
Hi, Is it possible to use just the compass function of MPU9250 on my Flip32 F4 board (airbotF4 target) which has MPU6000 gyro?Currently using hmc5883(builtin on GPS module) and would like to use better compass.

democles9308, 2018-08-19:
@tnako,
Did it work on your matek ? I adapted your patch for an OMNIBUSF4(PRO) board but the magnetometer is not recognised  when i set mpu9250 for th emagnetometer in the setup. As I would like to use the mpu9250's high address (0x69) by pulling SD0 to 3V3, I changed it in the common_hardware.c file 
BUSDEV_REGISTER_I2C(busdev_mpu9250,     DEVHW_MPU9250,      MPU9250_I2C_BUS,    0x69,        MPU_INT_EXTI,   DEVFLAGS_NONE);

Thx for the help

democles9308, 2018-08-20:
Hi @tnako,
Ol I'll have to investigate further... I2C works because OLED gives info...will look into the code and try to solve it. Thx anyway

democles9308, 2018-08-25:
Ok... got it to work. Dit some changes in the  OMNIBUSF4/4 target.h, common_hardware.c and the compass_ak8963.c file.
I' ve put the bypass code in the ak8963Detect () function and this made it working..
```
 bool ak8963Detect(magDev_t * mag)
{
    //Reset MPU9250 registers
    mag->busDev = busDeviceInit(BUSTYPE_I2C, DEVHW_MPU9250, mag->magSensorToUse, OWNER_COMPASS);
    busWrite(mag->busDev, 0x6B, 0x80);
    delay(100);
    // Set bypass mode
    busWrite(mag->busDev, 0x37, 0x02);
    delay(100);
    busDeviceDeInit(mag->busDev);
```

0zoon0, 2018-09-23:
Hi guys,
basically inspired by tnako work I've added a new driver compass_mpu925x_ak8963 which allows a MPU9250 or MPU9255 to be installed on I2C bus and setting bypass on it, we get an AK9863 compass instead.
Here are the commits:
https://github.com/0zoon0/inav/commit/e079fab3098087323cf0a6dae88f8990f10f423e
https://github.com/0zoon0/inav-configurator/commit/3cec949272150ac4b259caf4b080bf300c241eda

brightproject, 2023-07-11:
I read the entire branch, but did not understand much, due to insufficient knowledge in programming.
Perhaps someone can tell me what could be the cause of my [troubles](https://github.com/bolderflight/invensense-imu/issues/120) with `MPU9250`?
[image]