iNavFlight/inav #1761: Feature Request: BLHeli32 PassThrough and D-Shot (150 would be enough)

SoulPictures, 2017-06-04:
Hi there Dev. Team, the new BLHeli32 ESC got quiet neat features for Wings like Current Protection. Due to Betaflight already having Passthrough working (BF3.2) it shouldn't be hard to port it? Additionally an D-Shot mode, even the slowest would be perfect to completely eliminate noise on the ESC line. Are there possibilities to port those features?

digitalentity, 2017-09-08:
Very low priority. DSHOT doesn't give any benefit over "analog" protocol but actually masks a potential problem with EMI noise.

closed 2017-09-08 by digitalentity

kingdraki, 2017-09-16:
BLHeli32 ESC featuring onboard telemetry and can report VBAT and Current via UART to the FC. This would be a great benefit to use these Sensors in Inav.

VoicOfReason, 2017-10-27:
I agree

Feldsalat, 2017-12-23:
I had to flash betaflight today, only to adjust my 32bit esc. Would be nice if that would also be possible with Inav :-)

digitalentity, 2017-12-24:
@Feldsalat please check if https://github.com/iNavFlight/inav/pull/2530 works for you