iNavFlight/inav #1762: iNav 1.7.1 no ALT icon

wiesiekkr, 2017-06-04:
The icon should be at the top right
https://goo.gl/photos/pGCyv96kc6tyRvu36

wiesiekkr, 2017-06-04:
Board: KFC32F3

giacomo892, 2017-06-07:
You need a GPS fix

wiesiekkr, 2017-06-07:
Of course I have GPS fix......and 400m AGL :)

https://goo.gl/photos/xQFx9op5w1aQYwMy8

flounderscore, 2017-06-07:
As far as I know the current firmware doesn't show an icon for altitude.

marbalon, 2017-06-08:
Yes but should show, it is wrong symbol in definitions. Will be fixed in next release.

flounderscore, 2017-06-08:
I'm working on replacing the betaflight font with the default MWOSD font.
The symbol definition is going to be updated as well.

digitalentity, 2017-06-09:
Fix merged. @flounderscore how is your work coming along?

closed 2017-06-09 by digitalentity

flounderscore, 2017-06-09:
I updated the symbols file and osd.c a while back, but I haven't had time
to test it yet. Hopefully I'll get a chance this weekend (too many other
projects :-o ). Once I submit the PR I will need some help with the vario
as I've never used it and I'm not 100% sure how it's used.